Frogging-Family/nvidia-all #2: Cannot change MemoryTransferRate and GPUGraphicsClock Offsets using nvidia-all

NastySpaghetti, 2020-05-14:
For reference, here is [lshw output](https://pastebin.com/qwqkfj37)

Hello, so I wanted to use PowerMizer with nvidia-all but there is an issue. I can set up the preffered usage mode but I cannot change MemoryTransferRate and GPUGraphicsClock Offsets, it let me inputs numbers in the GUI but pressing enter doesn't register the change.

Moreover trying to modify the values using the terminal doen't work as well. [Here is the command outputs
](https://pastebin.com/ia3G3mLN)

[Here is the customization.cfg of nvidia-all](https://pastebin.com/zAiiP3HZ)
[Here the output of nvidia-bug-report](https://controlc.com/363ee132)

I tried putting cool-bits option to 0 using nvidia-xconfig then I built nvidia-all using cool-bits=0 in the customization.cfg. Then finally put cool-bits to 31 using nvidia-xconfig.Same issue
[Here is the customization.cfg of nvidia-all](https://pastebin.com/zN57YXZU)
[Here the output of nvidia-bug-report](https://controlc.com/4fe07015)

Tk-Glitch, 2020-05-14:
Thank you 😃 I'll be looking into it asap

NastySpaghetti, 2020-05-14:
I did one last test.
With the build installed using "cool-bits=0". I rebooted with Arch native 5.6 kernel.
[Here is the output of nvidia-bug-report](https://controlc.com/148d2aad).
Same issue.

Tk-Glitch, 2020-05-16:
Will reopen if needed.

closed 2020-05-16 by Tk-Glitch