Frogging-Family/nvidia-all #3: How to uninstall the drivers

Unser395, 2020-07-09:
Hello,
I'm interested in trying out the beta vulkan drivers, but I'd rather know how I am supposed to uninstall them before I do anything. Do you have any indications ?
I am on Manjaro i3.

Thank you very much.

Tk-Glitch, 2020-07-12:
Hello!
They are conflicting with Manjaro packages, so pacman will offer you to replace your current package both ways.

Trying to install from nvidia-all while having Manjaro packages installed? -> pacman will ask you if you want to replace Manjaro packages.
Trying to install Manjaro packages while having nvidia-all packages installed? -> pacman will ask you if you want to replace nvidia-all packages.

For the manual way with no replacement, just plain removal, something like `sudo pacman -R nvidia-* lib32-nbidia* opencl-nvidia-*` would do.

Unser395, 2020-07-14:
Thank you very much Tk-Glitch, no problem uninstalling the drivers with your solution.
I Hope you have a nice day !

closed 2020-07-14 by Unser395