Frogging-Family/nvidia-all #5: 440.100 driver fails to install

MrWissenschaft, 2020-08-19:
When installing 440.100 driver on 5.8.1-3-tkg-pds, DKMS produces an error. I have attached the terminal output and the make log for your perusal.

Thank you for all your hard work, and thank you in advance for looking into this for me (I only want to install it because the 450 series drivers make Final Fantasy XIV repeatedly crash lol)

[DKMS error 440-100.txt](https://github.com/Frogging-Family/nvidia-all/files/5098313/DKMS.error.440-100.txt)
[DKMS error 440-100 make log.txt](https://github.com/Frogging-Family/nvidia-all/files/5098314/DKMS.error.440-100.make.log.txt)
[image]

closed 2020-08-19 by Tk-Glitch

MrWissenschaft, 2020-08-19:
That's resolved the issue perfectly, thank you so much for your prompt response and resolution

Have a great day

Tk-Glitch, 2020-08-19:
You're welcome :) Enjoy!