Frogging-Family/nvidia-all #6: Prime render is broken?

donuts12, 2020-09-24:
I've tried your package install and everything was smooth and working but when i installed this nvidia-prime package from arch https://www.archlinux.org/packages/extra/any/nvidia-prime/ with prime-run command it doesn't work nor with optimus-manager https://github.com/Shatur95/optimus-manager-qt in hybrid mode using the lutris option "prime-rendering' selected. 

I've ran a test using the official nvidia drivers from a downgraded version for 450.66 for both dkms and normal kernel with nvidia-prime and both work fine. 

I'm using 970M Nvidia driver with Intel Broadwell with the TKG PDS Broadwell kernel and 450.66 tkg nvidia dkms package.

Tk-Glitch, 2020-09-25:
That's pretty weird. Could you please get a log of the error you get when it fails?

donuts12, 2020-09-25:
If I could find the logs for the error I would but the nvidia-prime package from Arch linux is only 150kb size and is stored in /usr/bin/prime-run 
#!/bin/bash
__NV_PRIME_RENDER_OFFLOAD=1 __VK_LAYER_NV_optimus=NVIDIA_only __GLX_VENDOR_LIBRARY_NAME=nvidia "$@" 
All you need is to run "prime-run" in terminal after install and you got prime activated for the nvidia card without having to install the optimus-manager and then just flip the prime offloading switch in lutris. This works for nvidia drivers from arch and would be nice addition if you add them into nvidia-all package as well. This was tested on my Msi Laptop i7 5700Hq, Nvidia 970M. I'm using vanilla arch linux with Plasma.

Tk-Glitch, 2020-10-08:
I won't have access to Nvidia hardware for a while, so I'll be unable to test that properly. I'm pretty surprised considering the glorious lack of changes for a long while on the Arch packages. A sudden breakage out of nowhere feels wrong, still. If anyone could provide a xorg log, in case it gives any clue on what could be wrong, I'd be thankful.

Tk-Glitch, 2020-11-21:
So, I was able to test on a 7700HQ/1050Ti laptop, and it worked fine. I'm don't know what could break for you here :/

closed 2026-08-18 by damachine