Frogging-Family/nvidia-all #8: How to uninstall?

Klairm, 2020-09-28:
How to uinstall drivers installed with this script?

openglfreak, 2020-09-28:
The same way other packages are uninstalled...

CodingKoopa, 2020-09-28:
You should be able to run `pacman -Qs nvidia`, pick out the `-tkg` packages, and run `pacman -Rs` for them.

Klairm, 2020-09-28:
I did that but for some reason Nvidia drivers still there

FLGX06, 2020-09-28:
Did you reboot? Those drivers are applied to kernel(s) using DKMS

Klairm, 2020-09-28:
Yep, I did reboot, guess I should reinstall the system

FLGX06, 2020-09-28:
Are you sure you are not using nouveau? Unofficial NVIDIA driver which is open source and has limitations (it is in kernel)

Klairm, 2020-09-28:
Well now I just trying to reinstall it (because I can't get video right now) but I get dkms install returning 10 status, what's should I do?

closed 2020-09-28 by Klairm

FLGX06, 2020-09-29:
Just for anyone wondering why @Klairm closed this, @Klairm was using old repo locally (didn't update) which had older drivers which apparently caused issues

Klairm, 2020-09-29:
> Just for anyone wondering why @Klairm closed this, @Klairm was using old repo locally (didn't update) which had older drivers which apparently caused issues

Yep, sorry for not specify that!