Frogging-Family/nvidia-all #9: Apparently im not the only one that had the issue...

Ohio2, 2020-10-06:
https://www.reddit.com/r/archlinux/comments/ed0d2m/i_fucked_up_my_xorg/
Just because of the nvidia-tkg drivers, i cannot start my XOrg on my linux-tkg and i think it should be both here and on the linux-tkg page.

Ohio2, 2020-10-06:
also to note; op in the reddit comments said that he did mkinitcpio -P i also did that!

Ohio2, 2020-10-06:
I also use arch.........

Ohio2, 2020-10-06:
Wait i also had the same fuckin log https://pastebin.com/0AzZZUxh
like literally.

Ohio2, 2020-10-06:
Also your driver works normally with Zen...?

Ohio2, 2020-10-22:
I fixed this issue??? i dunno 
[image]

closed 2020-10-22 by Ohio2

Tk-Glitch, 2020-10-22:
What am I looking at? :frog: