Frogging-Family/nvidia-all #12: AUR ?

ysolis, 2020-10-21:
Why not host the PKGBUILD in AUR ?

Ohio2, 2020-10-22:
would seem like an good idea, but you know... there exist AUR rules, and i think it's breaking it.

ronoverdrive, 2020-10-22:
> Why not host the PKGBUILD in AUR ?

The AUR disallows interactive scripts afaik. I don't know about cfg files, but I wouldn't be surprised if they're disallowed too. So because his scripts don't conform to their rules TkG is better off doing what he's been doing by self hosting.

telans, 2020-10-22:
I doubt the GPL 'patch' would be accepted on the AUR too

ronoverdrive, 2020-10-23:
> I doubt the GPL 'patch' would be accepted on the AUR too

I don't think that would be a problem. People apply unofficial patches all the time in AUR packages. The big thing is the interactivity because they want you to have to edit the PKGBUILD to change configuration options I think because it forces you to inspect the PKGBUILD for malicious code which they recommend to be standard practice.

telans, 2020-10-23:
It's not because it's a patch, it's because it circumvents licencing and copyrights