Frogging-Family/nvidia-all #13: Can't seem to install

LiamDawe, 2020-10-22:
Hi, tried to use it today to rollback to a previous driver.

>   -> Applying kernel-5.9.patch for 5.9.1-arch1-1...
> patching file common/inc/nv-mm.h
> Hunk #1 FAILED at 142.
> 1 out of 1 hunk FAILED -- saving rejects to file common/inc/nv-mm.h.rej
> patching file conftest.sh
> Hunk #2 FAILED at 2702.
> Hunk #3 succeeded at 2805 (offset 38 lines).
> Hunk #4 FAILED at 3873.
> 2 out of 4 hunks FAILED -- saving rejects to file conftest.sh.rej
> patching file nvidia-drm/nvidia-drm-drv.c
> Hunk #1 FAILED at 471.
> Hunk #2 FAILED at 717.
> 2 out of 2 hunks FAILED -- saving rejects to file nvidia-drm/nvidia-drm-drv.c.rej
> patching file nvidia-drm/nvidia-drm-gem.h
> Hunk #1 FAILED at 94.
> Hunk #2 FAILED at 143.
> 2 out of 2 hunks FAILED -- saving rejects to file nvidia-drm/nvidia-drm-gem.h.rej
> patching file nvidia-drm/nvidia-drm-utils.c
> Reversed (or previously applied) patch detected!  Assume -R? [n] 

I know there's certain issues with kernel 5.9, but I thought they were already sorted here? If you need other details let me know what to provide.

Tk-Glitch, 2020-10-22:
Can you please give me the version of the driver? Thanks!

LiamDawe, 2020-10-22:
Doh! Sorry, this was trying to install 450 series: 450.80.02

Tk-Glitch, 2020-10-22:
Thank you very much!

closed 2020-10-22 by Tk-Glitch