Frogging-Family/nvidia-all #34: Needs a patch for Linux Kernel 5.12.0

camdenorrb, 2021-04-03:
Tested Nvidia versions: [465, 460]

Fails to install DKMS Module:
```
(4/5) Install DKMS modules
==> dkms install --no-depmod -m nvidia -v 465.19.01 -k 5.12.0-rc5-144-tkg-cfs
Error! Bad return status for module build on kernel: 5.12.0-rc5-144-tkg-cfs (x86_64)
Consult /var/lib/dkms/nvidia/465.19.01/build/make.log for more information.
==> Warning, `dkms install --no-depmod -m nvidia -v 465.19.01 -k 5.12.0-rc5-144-tkg-cfs' returned 10
```

Make log error:
```
  CC [M]  /var/lib/dkms/nvidia/465.19.01/build/nvidia/procfs_nvswitch.o
/var/lib/dkms/nvidia/465.19.01/build/nvidia/nv-caps.c: In function ‘nv_cap_close_fd’:
/var/lib/dkms/nvidia/465.19.01/build/nvidia/nv-caps.c:598:5: error: implicit declaration of function ‘sys_close’ [-Werror=implicit-function-declaration]
  598 |     sys_close(fd);
      |     ^~~~~~~~~
  CC [M]  /var/lib/dkms/nvidia/465.19.01/build/nvidia-uvm/uvm_common.o
  CC [M]  /var/lib/dkms/nvidia/465.19.01/build/nvidia-uvm/uvm_linux.o
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/nvidia/465.19.01/build/nvidia/nv-caps.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1851: /var/lib/dkms/nvidia/465.19.01/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.12.0-rc5-144-tkg-cfs/build'
make: *** [Makefile:80: modules] Error 2
```

Arch post that references a similar issue along with providing a patch file: https://bbs.archlinux.org/viewtopic.php?id=262091&p=2

flgx16, 2021-04-03:
465.19.01 works fine with `5.12.0-rc5-144-tkg-cfs` on my computer

camdenorrb, 2021-04-03:
@flgx16 Any idea what I did differently?

flgx16, 2021-04-03:
> @flgx16 Any idea what I did differently?

I know I was being asked if I want to strip unused symbols when compiling linux-tkg and answered `N` to it, maybe you answered yes to that and that might be causing issues?

camdenorrb, 2021-04-03:
Seems to have done the trick, thank you :3

closed 2021-04-03 by camdenorrb