Frogging-Family/nvidia-all #35: Installation Doesn't Check for Linux-zen as a Valid Kernel

ThisNekoGuy, 2021-04-21:
Whenever I run the installation with the Linux-zen kernel installed, it doesn't check for whether my kernel is a valid kernel and claims I don't have the mainline linux kernel installed.

Tk-Glitch, 2021-04-22:
Are you talking about headers? You should be able to continue anyway on kernels with broken headers package (zen might be one of them). But the headers package for your kernel *must* be installed for sure, else the modules will fail to build.

ThisNekoGuy, 2021-05-23:
Sorry for responding late, I don't remember seeing a notification

Yeah, it didn't detect the Zen headers
I don't use the Zen kernel anymore, in favor of the tkg kernel now, but I remember it would warn me because I didn't have the regular kernel headers installed even though I had the Zen headers installed

It still worked, it was just odd being told that I didn't have headers installed when I very clearly did

Tk-Glitch, 2021-05-24:
It's just because the package is poorly made and doesn't respond to generic checks. The warning message was added explicitly for such cases.

closed 2026-08-18 by damachine