Frogging-Family/nvidia-all #37: Linux 5.12 - DKMS module isn't compiling

pkejval, 2021-04-28:
Logfile:

> DKMS make.log for nvidia-465.24.02 for kernel 5.12.0-150-tkg-bmq (x86_64)
St 28. dubna 2021, 09:14:23 CEST
make[1]: Vstupuje se do adresáře „/usr/lib/modules/5.12.0-150-tkg-bmq/build“
  SYMLINK /var/lib/dkms/nvidia/465.24.02/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/465.24.02/build/nvidia-modeset/nv-modeset-kernel.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-peermem/nvidia-ib-peermem-stub.o
  LD [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-ib-peermem-stub.o
 CONFTEST: set_pages_uc
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: acquire_console_sem
 CONFTEST: console_lock
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: sg_alloc_table
 CONFTEST: acpi_walk_namespace
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: efi_enabled
 CONFTEST: pde_data
 CONFTEST: proc_remove
 CONFTEST: pm_vt_switch_required
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_get_property
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: node_end_pfn
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_remove_bus_device
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: backlight_device_register
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write
 CONFTEST: kthread_create_on_node
 CONFTEST: of_find_matching_node
 CONFTEST: dma_direct_map_resource
 CONFTEST: dev_is_pci
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
 CONFTEST: flush_cache_all
 CONFTEST: vmf_insert_pfn
 CONFTEST: jiffies_to_timespec
 CONFTEST: ktime_get_raw_ts64
 CONFTEST: ktime_get_real_ts64
 CONFTEST: full_name_hash
 CONFTEST: hlist_for_each_entry
 CONFTEST: pci_enable_atomic_ops_to_root
 CONFTEST: vga_tryget
 CONFTEST: iterate_fd
 CONFTEST: pgprot_decrypted
 CONFTEST: seq_read_iter
 CONFTEST: unsafe_follow_pfn
[... 147 lines cut ...]
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia/nvlink_caps.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia/linux_nvswitch.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia/procfs_nvswitch.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/uvm_common.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/uvm_linux.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/nvstatus.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/nvCpuUuid.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/nv-kthread-q.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/nv-kthread-q-selftest.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/uvm.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/uvm_tools.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/uvm_global.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/uvm_gpu.o
/var/lib/dkms/nvidia/465.24.02/build/nvidia/nv-caps.c: In function ‘nv_cap_close_fd’:
/var/lib/dkms/nvidia/465.24.02/build/nvidia/nv-caps.c:598:5: error: implicit declaration of function ‘sys_close’ [-Werror=implicit-function-declaration]
  598 |     sys_close(fd);
      |     ^~~~~~~~~
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/uvm_gpu_isr.o
  CC [M]  /var/lib/dkms/nvidia/465.24.02/build/nvidia-uvm/uvm_procfs.o
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/nvidia/465.24.02/build/nvidia/nv-caps.o] Chyba 1
make[2]: *** Čeká se na nedokončené úlohy…
make[1]: *** [Makefile:1851: /var/lib/dkms/nvidia/465.24.02/build] Chyba 2
make[1]: Opouští se adresář „/usr/lib/modules/5.12.0-150-tkg-bmq/build“
make: *** [Makefile:80: modules] Chyba 2

Tk-Glitch, 2021-04-28:
It builds fine on my end. Make sure you have the kernel headers installed for your 5.12.0-150-tkg-bmq. `sys_close` is defined in the kernel headers so this shouldn't happen if you have them installed properly.

pkejval, 2021-04-28:
I have headers installed. But I now recall that I compiled kernel with new option "exclude unused kernel symbols". I'll try to recompile kernel with this option disabled and let you know.

pkejval, 2021-04-28:
Yes. It's working now! :)

closed 2021-04-28 by pkejval