Frogging-Family/nvidia-all #38: VK_ERROR_INITIALIZATION_FAILED With Unreal Engine 4 Games

d10sfan, 2021-04-29:
I encountered an issue when using a combination of linux-tkg kernel (in this case linux511-tkg-upds-5.11.17-152-x86_64) with nvidia drivers (tested on both 455.50.10 and 460.67).

With Unreal Engine 4 games, such as TO4 (https://discord.com/channels/155794864305471497/803665151102025778/837351849275293717), the game crashes with the following error.

LowLevelFatalError [File:Unknown] [Line: 794] Result failed, VkResult=-3 at D:/Build/++UE4/Sync/Engine/Source/Runtime/VulkanRHI/Private/VulkanSwapChain.cpp:493 with error VK_ERROR_INITIALIZATION_FAILED Signal 11 caught.

Other vulkan games, like Unity vulkan games, or open source engines, seem to work fine. Also, Beyond a Steel Sky, another UE4 game seems to work fine.

I am running Manjaro with a GTX 1070. If I use manjaro's 5.10 kernel with 460.73.01 then everything seems to work fine.

Tk-Glitch, 2021-04-29:
> tested on both 455.50.10 and 460.67
Why are you using these if 460.73.01 works with that game? The kernel itself doesn't make a difference as long as the driver supports it.

d10sfan, 2021-04-29:
Ah sorry didn't explain it well. I just tried tkg-installed 460.73.01 with the linux kernel installed with tkg and that failed with that error above, but trying the same nvidia tkg installed 460.73.01 with the manjaro linux kernel for 5.10.32-1 works fine.

I could try manjaro's version of 5.11 kernel if that'll help narrow it down.

d10sfan, 2021-04-29:
Sorry didn't see that. I tried using 460.73.01 from TKG and that actually gave me the same issue (running with either the tkg kernel or the normal 5.10 kernel). Using the manjaro installed 460.73.01 seems to work ok so far.

d10sfan, 2021-04-30:
Looks like this was related to some strange issue with something in my .config folder. Re-creating that caused the issue to disappear. I suppose something in there was corrupted.

Sorry for the confusion and thanks for the responses.

closed 2021-04-30 by d10sfan

Tk-Glitch, 2021-04-30:
Looks like some shader cache issue. That's pretty rare, but can happen from time to time. I'm glad you got it fixed!