Frogging-Family/nvidia-all #43: NVIDIA 470 on Arch

LiamDawe, 2021-06-22:
Hey, tried installing with the custom option for the new NVIDIA 470 driver.

Gets stuck here:

> ==> Starting package_nvidia-egl-wayland-tkg()...
> install: cannot stat 'libnvidia-egl-wayland.so.1.1.5': No such file or directory
> ==> ERROR: A failure occurred in package_nvidia-egl-wayland-tkg().

SimpliFly03, 2021-06-22:
You can work it around by setting `_eglwayland="false"`. But even if you succeed you will get `Error!  Build of nvidia-ib-peermem-stub.ko failed for` error during DKMS.

NoXPhasma, 2021-06-22:
To fix the egl version error, you need to edit the version from `1.1.5` to `1.1.7` in the [PKGBUILD](https://github.com/Frogging-Family/nvidia-all/blob/master/PKGBUILD#L1019) file. But as @SimpliFly03 said, it still fails at DKMS.

Tk-Glitch, 2021-06-22:
Fixed with https://github.com/Frogging-Family/nvidia-all/commit/cb47e2d616bfcc47970fd455f8a839116f5265de

closed 2021-06-22 by Tk-Glitch

LiamDawe, 2021-06-22:
Hey, got told there's some stuff in the wrong place.

We need to have 

_nvngx.dll and nvngx.dll setup in /usr/lib/nvidia/wine/ otherwise DLSS doesn't work.

Tk-Glitch, 2021-06-22:
Oh I thought it was in the lib dir after checking out the proton PR. Thanks for letting me know. I'll fix it asap.

Tk-Glitch, 2021-06-22:
Fixed btw - https://github.com/Frogging-Family/nvidia-all/commit/e00f0e0d634a21456eab12b4db43a9c71e9efffe