Frogging-Family/nvidia-all #44: DKMS fails on Kernel 5.12.10

rep-movsd, 2021-06-24:
DKMS fails for a custom Linux Kernel built from source:

```/var/lib/dkms/nvidia/465.31/build/nvidia/nv-caps.c: In function ‘nv_cap_close_fd’:
/var/lib/dkms/nvidia/465.31/build/nvidia/nv-caps.c:598:5: error: implicit declaration of function ‘sys_close’ [-Werror=implicit-function-declaration]
  598 |     sys_close(fd);
      |     ^~~~~~~~~
```

This happens even with the regular unpatched nvidia version BTW.
The script doesn't apply any patches at all.

But the stock kernel 5.12.* from Arch Linux works fine.

PrincessAkira, 2021-06-24:
Well i get a 
` dkms install --no-depmod -m nvidia -v 470.42.01 -k 5.12.12-169-tkg-pds-llvm
Error! Bad return status for module build on kernel: 5.12.12-169-tkg-pds-llvm (x86_64)
Consult /var/lib/dkms/nvidia/470.42.01/build/make.log for more information.
==> Warning, `dkms install --no-depmod -m nvidia -v 470.42.01 -k 5.12.12-169-tkg-pds-llvm' returned 10
`

Tk-Glitch, 2021-06-24:
@rep-movsd You need to install the headers for your custom kernel

@PrincessAkira You have chosen to build/install a llvm-built kernel, which will require you to force dkms to use llvm (which will break all your non-llvm kernels modules, and might not work for building nvidia modules ootb anyway). Nvidia systems shouldn't use llvm-built kernels for now imo, until dkms becomes a bit smarter at the very least.