Frogging-Family/nvidia-all #47: XNVCtrl library is not installed correctly.

BlueGoliath, 2021-07-04:
`XNVCtrl`, the library for `nvidia-settings` is not installed to `/usr/lib64`. As a result, any library that attempts to load the library, despite `nvidia-settings` working correctly, will fail.

If `nvidia-settings-tkg` is replaced via:

`sudo pacman -S nvidia-settings` 

then the `XNVCtrl` library is placed in /usr/lib64.

Tk-Glitch, 2021-07-04:
That library isn't supposed to be installed in `/usr/lib64`. Arch is patching it to have a shared version of it, which is not default. I'm fine porting the change over though.

Tk-Glitch, 2021-07-04:
Also, it isn't distributed in the nvidia driver packages and requires compiling it from source.

BlueGoliath, 2021-07-04:
Thanks, much appreciated.

Ubuntu also has this patch, strangely enough. I always thought it was shared since it has decently documented headers(located in /usr/include/NVCtrl).

anonish, 2021-07-05:
isnt there a package already in extra for this?
extra/libxnvctrl 465.31-1

BlueGoliath, 2021-07-05:
According to Pamac, libxnvctrl is a dependency of nvidia-settings. Also, while it isn't the same version as the driver(470 .42.01 vs 465.31).

anonish, 2021-07-06:
> According to Pamac, libxnvctrl is a dependency of nvidia-settings. Also, it isn't the same version as the driver(470 .42.01 vs 465.31).

i see what you mean.  weird that it's a separate package, but apparently (in arch anyway) it's nvidia-settings + libxnvctrl in the same PKGBUILD.  I was able to build to current beta version (470.42.01) by changing a couple lines ont he PKGBUILD (and ignoring the nvidia-settings package that it also builds).  the patch on arch seems to relate to installing the shared version of the library:
https://github.com/archlinux/svntogit-packages/blob/packages/nvidia-settings/trunk/libxnvctrl_so.patch

closed 2026-08-18 by damachine