Frogging-Family/nvidia-all #50: [Wayland] Windows Render Without Borders and Desktop UI is Tiny

ThisNekoGuy, 2021-07-23:
**Desktop Environment:** KDE Plasma 5.22.3
**GPU:** RTX 2080 Ti
**Driver Version:** 470.57.02 ("stable")

For some reason, I ended up with really weird rendering issues, despite the fact that 470 was supposed to be the driver release that brings proper Wayland support?

In Discord's case, it just refused to open properly entirely, [even with the config suggested by the Arch Wiki](https://wiki.archlinux.org/title/Wayland#Electron)

These screenshots were taken when I had the environment variable `QT_QPA_PLATFORM` set to `"xcb;wayland"`, [as per the Arch Wiki for suggests for Plasma,](https://wiki.archlinux.org/title/KDE#Kmail,_Kontact_and_Wayland) but these issues were still present with it set to [just `"wayland"`](https://wiki.archlinux.org/title/Wayland#Qt)

[image]
[image]
[image]

I also had SDDM set to run the session under Wayland:
```conf
sudo nano /etc/sddm.conf

[General]
DisplayServer=wayland
EnableHiDPI=true
```

Tk-Glitch, 2021-07-23:
I have faced none of those issues on that driver + kwin wayland on a 980ti and 1440p display. I was using a simpler setup than you are though, with only `KWIN_DRM_USE_EGL_STREAMS=1` set in an early boot script.

QT_QPA_PLATFORM was unset, and sddm was left default, just selecting the wayland session from the drop down menu.

ThisNekoGuy, 2021-07-25:
I reduced my environment variables and configuration to be that and the defaults and I still have the same result :/

Also, changing the Plasma compositor settings or app window borders causes the screen to irrecoverably turn black until a reboot

ThisNekoGuy, 2021-07-28:
@Tk-Glitch It also seemed to lower my max available framerate from 60hz to 30hz for some reason:

[image]

I also had the scale set to 200% back on Xorg but I take it it's trying to enforce HiDPI but is doing it wrong because I'm at an uncommon resolution or something? (3840x2160) Just a guess but Idk

Tk-Glitch, 2021-07-28:
Maybe? I know that some themes don't work or lead to huge issues on wayland, but your 30Hz lock looks totally unrelated. I don't have a 4k display to test, but I cannot reproduce any of your issues on my end with a 1440p display (DP) and a 1080p 120Hz display (DVI). Still on that 980Ti as I have no newer Nvidia GPU at hand.

ThisNekoGuy, 2021-07-28:
Ah, I didn't mention that I'm using a TV with HDMI, by the way; but Idk if that would impact it :/