Frogging-Family/nvidia-all #52: DKMS fails on Kernel 5.13.x

SM9CC, 2021-07-29:
```
  CC [M]  /var/lib/dkms/nvidia/455.50.19/build/nvidia-drm/nvidia-drm-modeset.o
/var/lib/dkms/nvidia/455.50.19/build/nvidia-drm/nvidia-drm-crtc.c:274:23: error: initialization of ‘int (*)(struct drm_plane *, struct drm_atomic_state *)’ from incompatible pointer type ‘int (*)(struct drm_plane>
  274 |     .atomic_check   = nv_drm_plane_atomic_check,
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/455.50.19/build/nvidia-drm/nvidia-drm-crtc.c:274:23: note: (near initialization for ‘nv_plane_helper_funcs.atomic_check’)
/var/lib/dkms/nvidia/455.50.19/build/nvidia-drm/nvidia-drm-crtc.c:275:23: error: initialization of ‘void (*)(struct drm_plane *, struct drm_atomic_state *)’ from incompatible pointer type ‘void (*)(struct drm_pla>
  275 |     .atomic_update  = nv_drm_plane_atomic_update,
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/455.50.19/build/nvidia-drm/nvidia-drm-crtc.c:275:23: note: (near initialization for ‘nv_plane_helper_funcs.atomic_update’)
/var/lib/dkms/nvidia/455.50.19/build/nvidia-drm/nvidia-drm-crtc.c:276:23: error: initialization of ‘void (*)(struct drm_plane *, struct drm_atomic_state *)’ from incompatible pointer type ‘void (*)(struct drm_pla>
  276 |     .atomic_disable = nv_drm_plane_atomic_disable,
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/455.50.19/build/nvidia-drm/nvidia-drm-crtc.c:276:23: note: (near initialization for ‘nv_plane_helper_funcs.atomic_disable’)
  CC [M]  /var/lib/dkms/nvidia/455.50.19/build/nvidia-drm/nvidia-drm-prime-fence.o
  CC [M]  /var/lib/dkms/nvidia/455.50.19/build/nvidia-drm/nvidia-drm-linux.o
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:273: /var/lib/dkms/nvidia/455.50.19/build/nvidia-drm/nvidia-drm-crtc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1851: /var/lib/dkms/nvidia/455.50.19/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.13.2-rt1-1-cacule-rdb-rt/build'
make: *** [Makefile:81: modules] Error 2
```

SM9CC, 2021-07-29:
There seems to be a patch which I shall try - https://forums.developer.nvidia.com/t/460-67-incompatible-with-kernel-5-13-at-the-moment/177543

Tk-Glitch, 2021-07-29:
455 series doesn't support 5.13, and we are not providing a patch for it since 465 and 470 do support it OOTB. Are you using 455 for a specific reason?

SM9CC, 2021-07-29:
i am using it for the Vulkan support, I was under the impression that the other versions did not have this?

Tk-Glitch, 2021-07-29:
396 onwards have had vulkan support. The vulkan developer drivers *usually* come with newer vulkan revisions support or features, but currently it's not worth it to use them over 470 as they lack some fixes and various new features such as xwayland acceleration. They also have not received updates for some time now.

SM9CC, 2021-07-29:
Thank you for clearing this up, I am now using 470 and its working perfectly.

closed 2021-08-15 by Tk-Glitch