Frogging-Family/nvidia-all #54: Why use this instead of arch packages?

geowarin, 2021-08-04:
Hello and thank you for your work!

Honest question here: why would I use these packages instead of the arch repos' one (nvidia, nvidia-lts, nvidia-dkms)?

From what I gathered, three reasons:

- This package is often updated more quickly than the official ones
- Using the vulkan dev drivers?
- This package contains patches for different kernels

About those patches. I have been unable to find information about what they do. I understand that some of them are compatibility patches for different kernels.

If it's only about kernel compatibility, why not contribute to the official packages instead?

But there might be more than this, if I read between the lines of this [other issue](https://github.com/Frogging-Family/nvidia-all/issues/12).

So what is the main difference for me if I don't have an exotic kernel, use the latest driver version and don't wish to use vulkan dev drivers?

Thanks!

Tk-Glitch, 2021-08-15:
> So what is the main difference for me if I don't have an exotic kernel, use the latest driver version and don't wish to use vulkan dev drivers?
None. The main goal of this beside giving quicker access to newer drivers is to be able to test different revisions easily in case of regression, or to simply be able to rollback to another version when things are breaking (which apparently happens a lot lately). If your usecase stays away from most issues and you don't need/care about exotic/development kernels nor vulkan dev drivers, there's no real point in using this.

gardotd426, 2022-01-15:
>If it's only about kernel compatibility, why not contribute to the official packages instead?

Because that's not up to him. Upstream would almost certainly not accept those patches. But that's not the point.

I mostly agree with TKG's sentiment, though I do think it's there is another reason to use this repo. Cloning it, setting `_dkms="true"` in `customization.cfg`, and running `makepkg -si` and selecting your kernel version installs *all* the components of the Nvidia driver. 

```
lib32-nvidia-utils-tkg 510.39.01-190
lib32-opencl-nvidia-tkg 510.39.01-190
nvidia-dkms-tkg 510.39.01-190
nvidia-egl-wayland-tkg 510.39.01-190
nvidia-settings-tkg 510.39.01-190
nvidia-utils-tkg 510.39.01-190
opencl-nvidia-tkg 510.39.01-190
```

Following most tutorials/instructions/etc you'd only end up with `nvidia` or `nvidia-dkms`, `nvidia-settings`, and maybe `nvidia-utils`. I've seen countless people on reddit where they say they followed the Arch Wiki or Lutris's GH instructions for installing the drivers, only to find out they don't have `lib32-nvidia-utils`, which means they have no lib32 vulkan driver (or even not having `nvidia-utils`, so having no vulkan driver period). 

This repo eliminates any of that, and ensures you have everything you need from the Nvidia driver. 

But I would say this issue can be closed, as it's not a real issue anyway and belongs more as a question on the discord, but regardless it's already been answered.