Frogging-Family/nvidia-all #55: Rudimentary clang / llvm support?

flindeberg, 2021-08-07:
There is circumstantial evidence which suggests that it might take a while for DKMS to natively support clang (i.e., https://github.com/dell/dkms/issues/124), however there are others which have managed to get DKMS nvidia-drivers working (i.e., https://rglinuxtech.com/?p=2903, they have messed up their certificates though, so proceed with care) without issues. 

The reason I find this (partially) worthwhile is that a clang + LTO compiled linux kernel gives about a percentage point of performance compared to a gcc-compiled kernel (i.e., https://www.phoronix.com/scan.php?page=article&item=clang-lto-kernel&num=3). And free performance is good, right? :-)

ptr1337, 2021-08-07:
These are literally 3 lines of code to change.

Snoop05, 2021-08-07:
It wouldn't make much sense to incorporate whatever compiler detection logic within the driver package as it wouldn't help you with other dkms packages on your system, therefore this really should be targeted at dell/dkms. If you want to use it on your system its quite simple: just append `export LLVM=1 CC=clang` to `/etc/dkms/framework.conf` - assuming you only have clang compiled kernels, if you have gcc as well then as ptr1337 mentioned, its trivial change in the dkms script - either check the kernel binary or the kernel config within its headers directory and export those vars depending on the result. I also noticed that other seem to spam tons of vars, for me this was all that was required to make it work (and the CC was only needed for nvidia's conftest)
I guess it may be a good idea to note this in the readme of both nvidia-all and linux-tkg repos 🙂

ptr1337, 2021-08-07:
There are also other ways, how to get this also work with the PKGBUILD, but that would be to much all in all.

timocapa, 2021-08-21:
I don't think the module compiles against an LTO'd clang kernel anyway.

```
  MODPOST /var/lib/dkms/nvidia/470.63.01/build/Module.symvers
ERROR: modpost: "__fentry__" [/var/lib/dkms/nvidia/470.63.01/build/nvidia-peermem.ko] undefined!
ERROR: modpost: "__sw_hweight64" [/var/lib/dkms/nvidia/470.63.01/build/nvidia-drm.ko] undefined!
ERROR: modpost: "__put_user_4" [/var/lib/dkms/nvidia/470.63.01/build/nvidia-drm.ko] undefined!
ERROR: modpost: "memmove" [/var/lib/dkms/nvidia/470.63.01/build/nvidia-drm.ko] undefined!
ERROR: modpost: "memcpy" [/var/lib/dkms/nvidia/470.63.01/build/nvidia-drm.ko] undefined!
ERROR: modpost: "memset" [/var/lib/dkms/nvidia/470.63.01/build/nvidia-drm.ko] undefined!
ERROR: modpost: "__x86_indirect_thunk_r11" [/var/lib/dkms/nvidia/470.63.01/build/nvidia-drm.ko] undefined!
ERROR: modpost: "__stack_chk_fail" [/var/lib/dkms/nvidia/470.63.01/build/nvidia-drm.ko] undefined!
ERROR: modpost: "__fentry__" [/var/lib/dkms/nvidia/470.63.01/build/nvidia-drm.ko] undefined!
ERROR: modpost: "fput" [/var/lib/dkms/nvidia/470.63.01/build/nvidia-drm.ko] undefined!
WARNING: modpost: suppressed 684 unresolved symbol warnings because there were too many)
make[2]: *** [scripts/Makefile.modpost:150: /var/lib/dkms/nvidia/470.63.01/build/Module.symvers] Error 1
make[2]: *** Deleting file '/var/lib/dkms/nvidia/470.63.01/build/Module.symvers'
make[1]: *** [Makefile:1777: modules] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.13.12-192-tkg-cacule-llvm/build'
make: *** [Makefile:80: modules] Error 2
```

timocapa, 2021-08-22:
Nevermind, second try.

```
timo@mauve ~> sudo dmesg | grep clang; lspci -nnk | grep "GTX\|nvidia"
[    0.000000] Linux version 5.13.12-192-tkg-cacule-llvm (linux513-tkg-cacule-llvm@archlinux) (ClangBuiltLinux clang version 14.0.0 (https://github.com/llvm/llvm-project d2bc4fa3c70ad9dd1723b2e75c9059c5db2052cf), LLD 14.0.0) #1 TKG SMP PREEMPT Sun, 22 Aug 2021 05:45:06 +0000
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP104 [GeForce GTX 1080] [10de:1b80] (rev a1)
	Kernel driver in use: nvidia
	Kernel modules: nvidia_drm, nvidia
```

ErdeFB, 2021-09-08:
> Nevermind, second try.
>
> ```
> timo@mauve ~> sudo dmesg | grep clang; lspci -nnk | grep "GTX\|nvidia"
> [    0.000000] Linux version 5.13.12-192-tkg-cacule-llvm (linux513-tkg-cacule-llvm@archlinux) [...]

How'd you solve this?

timocapa, 2021-09-08:
@ErdeFB as it has been mentioned further up, 

export LLVM=1 CC=clang to /etc/dkms/framework.conf and then it just started working for me

flindeberg, 2021-09-09:
@ErdeFB https://github.com/dell/dkms/issues/124 (the dkms-repo issue) also contains som other ways of modifying DKMS itself to pick the right compile tools. But `export LLVM=1 CC=clang` is undoubtedly easier and quicker to just get it to work.

Cabopust, 2021-11-20:
This seems to be solved in upstream dkms thanks to the merge -  https://github.com/dell/dkms/pull/169

timocapa, 2021-11-20:
Unfortunately building the nvidia-dkms module seems broken as of Clang 13 (and 14 for a while) anyway.

flindeberg, 2021-11-23:
Seem to work fine with latest DKMS (=> 3.0.2), with the fixes from https://github.com/dell/dkms/pull/169 . 

Here after a rebuild of a llvm-kernel on one of my machines using https://github.com/Frogging-Family/nvidia-all, https://github.com/Frogging-Family/linux-tkg and DKMS from arch-repos:
```
(3/6) Install DKMS modules
==> dkms install --no-depmod nvidia/495.44 -k 5.15.4-221-tkg-cacule-llvm
==> depmod 5.15.4-221-tkg-cacule-llvm
(4/6) Cleaning up...
```
I'll close this in a day or two if I don't find any issues with Nvidia and llvm.

flindeberg, 2021-11-24:
I've started games with dxvk, vulkan and vkd3d with both llvm and gcc kernels with the nvidia drivers installed by nvidia-all and it all seems to work. 

I'm closing this issue with reference to https://github.com/dell/dkms/pull/169.

closed 2021-11-24 by flindeberg