Frogging-Family/nvidia-all #58: nvidia-settings icon is not visible for KDE

ranplayer, 2021-08-12:
After this commit https://github.com/Frogging-Family/nvidia-all/commit/05233ee23db6ad3f0782d1766459c09dddd9b1c2, the **nvidia-settings** icon is not visible for KDE. Reverting it locally made the icon be displayed again. I haven't opened a PullRequest because I'm not sure about the other desktop environments.

enveeed, 2021-08-13:
I have the same issue on gnome, despite using an icon theme which includes an nvidia-settings icon

Tk-Glitch, 2021-08-15:
Since this looks like it breaks almost universally, I have reverted the offending commit.

closed 2021-08-15 by Tk-Glitch

Cabopust, 2021-08-15:
Really sorry about that. I didn't test this well enough, which caused me to miss that you need to change not __PIXMAP_PATH__ to nvidia-settings, but the full __PIXMAP_PATH__/nvidia-settings.png. At the same time when I manually edited Icon to Icon=nvidia-settings everything worked.

Here is the corrected version which works correctly in all desktop environments. @ranplayer, @enveeed  Please try this version.

```
diff --git a/PKGBUILD b/PKGBUILD
index 40975a9..161189e 100644
--- a/PKGBUILD
+++ b/PKGBUILD
@@ -1296,7 +1296,7 @@ nvidia-settings-tkg() {
     install -D -m644 nvidia-settings.1.gz    -t "${pkgdir}/usr/share/man/man1"
     install -D -m644 nvidia-settings.png     -t "${pkgdir}/usr/share/pixmaps"
     install -D -m644 nvidia-settings.desktop -t "${pkgdir}/usr/share/applications"
-    sed -e 's:__UTILS_PATH__:/usr/bin:' -e 's:__PIXMAP_PATH__:/usr/share/pixmaps:' -i "${pkgdir}/usr/share/applications/nvidia-settings.desktop"
+    sed -e 's:__UTILS_PATH__:/usr/bin:' -e 's:__PIXMAP_PATH__/nvidia-settings.png:nvidia-settings:' -i "${pkgdir}/usr/share/applications/nvidia-settings.desktop"

     install -D -m755 "libnvidia-gtk3.so.${pkgver}" -t "${pkgdir}/usr/lib"
```