Frogging-Family/nvidia-all #60: Error in package_nvidia-dev-utils-tkg()

pkejval, 2021-09-17:
When I'm building from path "/home/<user>/source/nvidia-all" I got error:
`ln: failed to create symbolic link '/home/<user>.so': Permission denied
`

But when I move git repo from "/home/<user>/source/nvidia-all" to "/home/<user>/nvidia-all" build is working. I was building from my source folder for months now it's not working from it.

sogens, 2021-09-17:
having the exact same issue...

jnines, 2021-09-20:
Same issue with 470.74, had the same issue with the previous version.  The issue appears to be with [3e05cde](https://github.com/Frogging-Family/nvidia-all/commit/07d2aec324749dc62bdce57c4870dc3ca0d91328).  Commenting out the vulkan-producer block allows for compile/install.  Obviously not useful if you require whatever that does (CUDA maybe?).

jnines, 2021-09-21:
So I _maybe_ found the issue.  On line 330 of the PKGBUILD there's a function
```
_create_links() {
  # create missing soname links
  for _lib in $(find "$pkgdir" -name '*.so*' | grep -v 'xorg/'); do
    # Get soname/base name
    _soname=$(dirname "$_lib")/$(readelf -d "$_lib" | grep -Po 'SONAME.*: \[\K[^]]*' || true)
    _base=$(echo "$_soname" | sed -r 's/(.*).so.*/\1.so/')

    # Create missing links
    [ -e "$_soname" ] || ln -s $(basename "$_lib") "$_soname"
    [ -e "$_base" ] || ln -s $(basename "$_soname") "$_base"
  done
}
```
This is called at the end of each grouping.  It runs `readelf -d` and greps for `SONAME` in the results.  For whatever reason (not my area of expertise) `libnvidia-vulkan-producer` doesn't have a `SONAME` entry listed, so it appears it errors out.  I just commented it out as I don't install egl_wayland and it appears that it requires it `(NEEDED)             Shared library: [libnvidia-egl-wayland.so.1]`.  You could just change it to
```
_create_links() {
  # create missing soname links
  for _lib in $(find "$pkgdir" -name '*.so*' | grep -v 'xorg/'); do
    # Get soname/base name
    if [[ $_lib != *libnvidia-vulkan-producer.* ]]; then
      _soname=$(dirname "$_lib")/$(readelf -d "$_lib" | grep -Po 'SONAME.*: \[\K[^]]*' || true)
      _base=$(echo "$_soname" | sed -r 's/(.*).so.*/\1.so/')

      # Create missing links
      [ -e "$_soname" ] || ln -sv $(basename "$_lib") "$_soname"
      [ -e "$_base" ] || ln -sv $(basename "$_soname") "$_base"
    fi
  done
}
```
Probably something off there but...
This driver also seems to cause a small issue with sddm.

Tk-Glitch, 2021-09-21:
While I cannot reproduce the issue on my end, @jnines is right. That file doesn't have a SONAME entry. I'm fine with the proposed workaround.

closed 2021-09-21 by Tk-Glitch