Frogging-Family/nvidia-all #61: Fails to load on 5.14.9 Clang LTO TKG Kernel

C43H66N12O12S2, 2021-10-02:
Both the DKMS and regular flavor of both the stable branch and vulkan dev branch fails to load with this message:

```
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_rax (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_r8 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_rdx (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_rcx (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_r10 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_r9 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_rdi (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_r11 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_rsi (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_rbx (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_jmp_r12 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_rax (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_r8 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_rbx (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_r13 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_r9 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_rdx (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_rcx (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_r10 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_r12 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_r11 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_r15 (err -2)
kernel: nvidia: Unknown symbol __x86_indirect_alt_call_r14 (err -2)
```

The very same package works just fine with 5.14.8 TKG (also Clang and LTO).

The regular version building fails with "different compilers" error, even though both the kernel and the module is being built by the exact same compiler and version, so maybe that's a clue? 

I'm happy to provide any additional information required.

Tk-Glitch, 2021-10-02:
Clang built kernels aren't really supported, even less so with Nvidia

flindeberg, 2021-11-24:
@C43H66N12O12S2 Can you verify that this still is the case with latest DKMS? (i.e. DKMS >= 3.0.2)

C43H66N12O12S2, 2021-12-24:
@flindeberg I haven't tested it, but I doubt it's fixed. To be clear, I highly doubt this is a Clang issue and it's definitely not a DKMS issue, as I've stated in the OP, normal kernel modules (without DKMS) failed with the same error. This seems to be more of a quirk of the nvidia driver, and it's dependent on one setting in the kernel configuration, disabling that solves it for DKMS/Non-DKMS, Clang and so on.

The reason I've left this issue open is to help anybody else who encountered the same issue. Otherwise, this issue is actually resolved.