Frogging-Family/nvidia-all #71: Unable to install the newest 495.44 driver

LiamDawe, 2021-11-01:
Hi, trying to install the newer driver and it seems to fail here:

> ==> Starting package_nvidia-utils-tkg()...
> readelf: Error: '/home/liam/.local/share/Trash/files/nvidia-all' is not an ordinary file
> readelf: Error: '(1)/nvidia-all/pkg/nvidia-utils-tkg/usr/lib/libcuda.so.495.44': No such file
> ln: failed to create symbolic link '(1)/nvidia-all/pkg/nvidia-utils-tkg/usr/lib/': No such file or directory
> ==> ERROR: A failure occurred in package_nvidia-utils-tkg().
>     Aborting...
>   -> exit cleanup done
>   -> exit cleanup done

Full log: https://gist.github.com/LiamDawe/ec7e94a1a20243b7c7bc2c554905c125

maklamj, 2021-11-01:
Have you tried building in "~/"?
https://github.com/Frogging-Family/nvidia-all#how-to-run-the-installer

maklamj, 2021-11-01:
Have you installing building from "~/" path?
https://github.com/Frogging-Family/nvidia-all#how-to-run-the-installer

Tk-Glitch, 2021-11-01:
Fix incoming!

closed 2021-11-01 by Tk-Glitch