Frogging-Family/nvidia-all #76: Building the DKMS version on clang >= 13 fails

keksbg, 2021-11-21:
It appears that on clang >= 13 it refuses to build due to the `-mrecord-mcount` flag not existing.
Excerpt from `make.log`:
```
# [...]
 CONFTEST: ib_peer_memory_symbols
  CC [M]  /var/lib/dkms/nvidia/495.44/build/nvidia/nv.o
  CC [M]  /var/lib/dkms/nvidia/495.44/build/nvidia/nv-pci.o
  CC [M]  /var/lib/dkms/nvidia/495.44/build/nvidia/nv-acpi.o
  CC [M]  /var/lib/dkms/nvidia/495.44/build/nvidia/nv-cray.o
  CC [M]  /var/lib/dkms/nvidia/495.44/build/nvidia/nv-dma.o
  CC [M]  /var/lib/dkms/nvidia/495.44/build/nvidia/nv-i2c.o
  CC [M]  /var/lib/dkms/nvidia/495.44/build/nvidia/nv-mmap.o
  CC [M]  /var/lib/dkms/nvidia/495.44/build/nvidia/nv-p2p.o
clang-13: error: unsupported option '-mrecord-mcount' for target 'x86_64-pc-linux-gnu'
clang-13: error: unsupported option '-mrecord-mcount' for target 'x86_64-pc-linux-gnu'
make[2]: *** [scripts/Makefile.build:272: /var/lib/dkms/nvidia/495.44/build/nvidia/nv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
# [...]
```
Downgrading to clang 12 makes it work again with no errors.

ptr1337, 2021-11-21:
For me its working, also with clang-14. 
But i dont use atm -tkg nvidia drivers. 

You can try to built with this patch dkms, then remove out of your framework.conf the exports.
```
From 3eb09af9358a1bd55b9ca04866b5161e714bd5f2 Mon Sep 17 00:00:00 2001
From: Simone Caronni <negativo17@gmail.com>
Date: Tue, 16 Nov 2021 16:53:56 +0100
Subject: [PATCH 1/5] Add more deprecated options

---
 dkms.in | 10 +++++++++-
 1 file changed, 9 insertions(+), 1 deletion(-)

diff --git a/dkms.in b/dkms.in
index 70b26ac..4d54378 100644
--- a/dkms.in
+++ b/dkms.in
@@ -2200,9 +2200,17 @@ while (($# > 0)); do
             exit 0
             ;;
         --no-initrd)
-            # This is a old option, consume and warn
+            # This is an old option, consume and warn
             deprecated $"--no-initrd"
             ;;
+        --no-clean-kernel)
+            # This is an old option, consume and warn
+            deprecated $"--no-clean-kernel"
+            ;;
+        --no-prepare-kernel)
+            # This is an old option, consume and warn
+            deprecated $"--no-prepare-kernel"
+            ;;
         --binaries-only)
             binaries_only="binaries-only"
             ;;
-- 
2.34.0

From f5051d0543aaf933874ae512f5624ac1fd95056d Mon Sep 17 00:00:00 2001
From: Nicolas Iooss <nicolas.iooss_git@polytechnique.org>
Date: Wed, 17 Nov 2021 09:50:32 +0000
Subject: [PATCH 2/5] Enable using add, build and install with a directory
 again

Since commit e73f9bbf2013 ("dkms: fixup add/remove...uninstall
permission checking"), it was no longer possible to use "dkms add" with
a directory:

    $ dkms add test/dkms_test-1.0
    Error! Arguments <module> and <module-version> are not specified.
    Usage: add <module>/<module-version> or
           add -m <module>/<module-version> or
           add -m <module> -v <module-version>

But when options -m and -v are given, the source directory is ignored.

Fix this issue by removing the check for -m and -v in add, build and
install actions. This makes the following commands work again (for
example on Arch Linux):

    $ dkms add test/dkms_test-1.0
    Creating symlink /var/lib/dkms/dkms_test/1.0/source -> /usr/src/dkms_test-1.0

    $ dkms remove -m dkms_test -v 1.0 --all
    ...
    $ dkms install -k 5.15.2-arch1-1 test/dkms_test-1.0
    Creating symlink /var/lib/dkms/dkms_test/1.0/source -> /usr/src/dkms_test-1.0

[... 648 lines cut ...]
diff --git a/dkms.in b/dkms.in
index 9c98a89..97cc1c9 100644
--- a/dkms.in
+++ b/dkms.in
@@ -571,6 +571,16 @@ read_conf()
         make_command="${MAKE[$index]}"
     done
 
+    # Check if clang was used to compile or lld was used to link the kernel.
+    if [[ -e $kernel_source_dir/vmlinux ]]; then
+      if  readelf -p .comment $kernel_source_dir/vmlinux | grep -q clang; then
+        make_command="${make_command} CC=clang"
+      fi
+      if  readelf -p .comment $kernel_source_dir/vmlinux | grep -q LLD; then
+        make_command="${make_command} LD=ld.lld"
+      fi
+    fi
+
     # Use the generic make and make clean commands if not specified
     [[ ! $make_command ]] && make_command="make -C $kernel_source_dir M=$dkms_tree/$module/$module_version/build"
     [[ ! $clean ]] && clean="make -C $kernel_source_dir M=$dkms_tree/$module/$module_version/build clean"
-- 
2.34.0

```

ptr1337, 2021-11-21:
Can be closed. is fixed with the latest dkms @archlinux repos

keksbg, 2021-11-21:
Yes, this does appear to have been fixed by the update to dkms.

closed 2021-11-21 by keksbg