Frogging-Family/nvidia-all #83: Compiling driver version 396.18 is impossible

DAC324, 2021-12-23:
Hello,
The heading of this package claims that it is possible to build drivers from version 396 up to the latest version. 
Unfortunately, compiling driver 396.18 fails:
```
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.c:88:1: warning: data definition has no type or storage class
   88 | NV_DEFINE_PROCFS_SINGLE_FILE(permissions);
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.c:88:1: error: type defaults to 'int' in declaration of 'NV_DEFINE_PROCFS_SINGLE_FILE' [-Werror=implicit-int]
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.c:88:1: warning: parameter names (without types) in function declaration
In file included from /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/common/inc/nv-linux.h:299,
                 from /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.c:30:
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.c: In function 'nvlink_permissions_init':
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/common/inc/nv-procfs.h:60:47: error: 'nv_procfs_permissions_fops' undeclared (first use in this function)
   60 |         const struct file_operations *fops = &nv_procfs_##__name##_fops;        \
      |                                               ^~~~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.c:108:26: note: in expansion of macro 'NV_CREATE_PROC_FILE'
  108 |     nvlink_permissions = NV_CREATE_PROC_FILE("permissions",
      |                          ^~~~~~~~~~~~~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/common/inc/nv-procfs.h:60:47: note: each undeclared identifier is reported only once for each function it appears in
   60 |         const struct file_operations *fops = &nv_procfs_##__name##_fops;        \
      |                                               ^~~~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.c:108:26: note: in expansion of macro 'NV_CREATE_PROC_FILE'
  108 |     nvlink_permissions = NV_CREATE_PROC_FILE("permissions",
      |                          ^~~~~~~~~~~~~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/common/inc/nv-procfs.h:61:19: error: 'const struct file_operations' has no member named 'proc_write'; did you mean 'splice_write'?
   61 |         if (fops->proc_write != 0)                                       \
      |                   ^~~~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.c:108:26: note: in expansion of macro 'NV_CREATE_PROC_FILE'
  108 |     nvlink_permissions = NV_CREATE_PROC_FILE("permissions",
      |                          ^~~~~~~~~~~~~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/common/inc/nv-procfs.h:63:60: error: passing argument 4 of 'proc_create_data' from incompatible pointer type [-Werror=incompatible-pointer-types]
   63 |         __entry = proc_create_data(filename, mode, parent, fops, __data);\
      |                                                            ^~~~
      |                                                            |
      |                                                            const struct file_operations *
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.c:108:26: note: in expansion of macro 'NV_CREATE_PROC_FILE'
  108 |     nvlink_permissions = NV_CREATE_PROC_FILE("permissions",
      |                          ^~~~~~~~~~~~~~~~~~~
In file included from ./include/linux/efi.h:19,
                 from /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/common/inc/nv-linux.h:205,
                 from /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.c:30:
./include/linux/proc_fs.h:106:48: note: expected 'const struct proc_ops *' but argument is of type 'const struct file_operations *'
  106 |                                                const struct proc_ops *,
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:280: /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nvlink_linux.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_api.h:30,
                 from /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_global.c:24:
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_va_range.h: In function 'uvm_va_range_vma_check':
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_va_range.h:670:45: error: 'struct mm_struct' has no member named 'mmap_sem'; did you mean 'mmap_base'?
  670 |     uvm_assert_mmap_sem_locked(&vma->vm_mm->mmap_sem);
      |                                             ^~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_lock.h:492:14: note: in definition of macro 'uvm_assert_mmap_sem_locked_mode'
  492 |       typeof(mmap_sem) _sem = (mmap_sem);                                           \
      |              ^~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_va_range.h:670:5: note: in expansion of macro 'uvm_assert_mmap_sem_locked'
  670 |     uvm_assert_mmap_sem_locked(&vma->vm_mm->mmap_sem);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_va_range.h:670:45: error: 'struct mm_struct' has no member named 'mmap_sem'; did you mean 'mmap_base'?
  670 |     uvm_assert_mmap_sem_locked(&vma->vm_mm->mmap_sem);
      |                                             ^~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_lock.h:492:32: note: in definition of macro 'uvm_assert_mmap_sem_locked_mode'
  492 |       typeof(mmap_sem) _sem = (mmap_sem);                                           \
      |                                ^~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_va_range.h:670:5: note: in expansion of macro 'uvm_assert_mmap_sem_locked'
  670 |     uvm_assert_mmap_sem_locked(&vma->vm_mm->mmap_sem);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_lock.h:29,
                 from /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia-uvm/uvm8_api.h:30,
[... 661 lines cut ...]
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nv-procfs.c:850:17: note: in expansion of macro 'NV_CREATE_PROC_FILE'
  850 |         entry = NV_CREATE_PROC_FILE("unbindLock", proc_nvidia_gpu, unbind_lock, nv);
      |                 ^~~~~~~~~~~~~~~~~~~
/home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/common/inc/nv-procfs.h:63:60: error: passing argument 4 of 'proc_create_data' from incompatible pointer type [-Werror=incompatible-pointer-types]
   63 |         __entry = proc_create_data(filename, mode, parent, fops, __data);\
[... cut ...]
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:280: /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO/nvidia/nv-procfs.o] Error 1
make[1]: *** [Makefile:1822: /home/gerd/Downloads/nvidia-390xx/nvidia-all/src/NVIDIA-Linux-x86_64-396.18/kernel-5.10.85-1-MANJARO] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.10.85-1-MANJARO/build'
make: *** [Makefile:79: modules] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

Tk-Glitch, 2021-12-23:
396 series requires kernel 5.3 or older, 5.1 being the recommended version for best compatibility. 5.10 is too new.

DAC324, 2021-12-23:
> 396 series requires kernel 5.3 or older, 5.1 being the recommended version for best compatibility. 5.10 is too new.
According to `PKGBUILD`, the latest kernel to work with the `396` driver series appears to be `5.7`:
```
if (( $(vercmp "$_kernel" "5.7") >= 0 )); then
      _kernel57="1"
      _whitelist57=( 396* 410* 415* 418* 430* 435* 440* )
    fi          
```
But the patches for kernels up to `5.7` also do not work flawlessly. There are many hunks rejected by `patch`.

Tk-Glitch, 2021-12-24:
The patch used to apply on previous 396 series drivers but it didn't result in working modules afaik. It was added to the allowed list only because the patch applied (likely on some vulkan dev build that supported transform feedback, more recent than mainline .18), and while I remember getting things to work on 5.3, it was never tested successfully on initial 5.4 release and never tested at all beyond that. After 150+ revisions of 5.4 at this point, things may be even deeper frogged now than at release.. I have added some notes in the initial selector when you call makepkg to clarify what should be expected compatibility wise, especially on the older versions, but ultimately those dinosaur drivers should be removed altogether as they serve no real purpose.

DAC324, 2021-12-24:
> but ultimately those dinosaur drivers should be removed altogether as they serve no real purpose.

I don't think so. There are quite a lot of older computers equipped with Nvidia Fermi GPUs, and why should we toss them in the bin if they are still working?
I am currently writing this on an ancient Lenovo ThinkCentre M91p, equipped with a Geforce GTX 560 card. 

Interestingly enough, this card runs happily with the even older driver `390`, under the latest kernel `5.16-rc6` :)

Tk-Glitch, 2021-12-24:
That's the thing. We're are not supporting legacy hardware because 340 and 390 are done for that, are still maintained by Nvidia, and are working well. 396 is and had been abandoned for a long while now.

DAC324, 2022-01-07:
> The patch used to apply on previous 396 series drivers but it didn't result in working modules afaik.

Tk-Glitch gave an important hint. Indeed, it looks like some of the updates required for kernels beyond **5.4** have been applied to the **390** driver, rather than to **396**. 
Thus, we have the somewhat weird situation that the source files for **390** are newer than those for **396** despite of their version number being higher. 

I just did an experiment: Extracted the **390.147** archive, applied the usual patch to the sources to have them running with kernel **5.16**, and then copied the entire `kernel` directory over my **396.18**  sources, thus replacing all of them. 
As a result, it was possible to build **396.18** modules against the kernel **5.16** headers, and they apparently even loaded. 

Unfortunately, Nvidia fails to properly maintain their online documentation. For example, the list of supported products for the **396.18** driver on [https://www.nvidia.com/Download/driverResults.aspx/133571/en-us](url) mentions the GeForce GTX 560 to be supported:
[quote]
**GeForce 500 Series:**
GeForce GTX 590, GeForce GTX 580, GeForce GTX 570, GeForce GTX 560 Ti, GeForce GTX 560 SE, **GeForce GTX 560**, GeForce GTX 555, GeForce GTX 550 Ti, GeForce GT 545, GeForce GT 530, GeForce GT 520, GeForce 510
[/quote]

But the driver itself refuses to load:
[quote]
[    30.891] (WW) NVIDIA(0): The NVIDIA **GeForce GTX 560 GPU** installed in this system is
[    30.891] (WW) NVIDIA(0):     supported through the NVIDIA 390.xx Legacy drivers. Please
[    30.891] (WW) NVIDIA(0):     visit http://www.nvidia.com/object/unix.html for more
[    30.891] (WW) NVIDIA(0):     information.  The 396.18 NVIDIA driver will ignore this
[    30.891] (WW) NVIDIA(0):     GPU.  Continuing probe... 
[    30.891] (EE) No devices detected.
[/quote]