Frogging-Family/nvidia-all #90: Error: "`nvidia.ko` not found"

ThisNekoGuy, 2022-02-01:
I was trying to run the following as a test for a personal script I was going to write later:
`modinfo /usr/lib/modules/$(uname -r)/kernel/drivers/video/nvidia.ko | grep ^version`

When I unexpectedly ran into an error:
`modinfo: ERROR: Module /usr/lib/modules/5.14.21-239-tkg-pds-llvm/kernel/drivers/video/nvidia.ko not found.`

I run `linux-tkg`, and I'm running the driver right now as I'm typing this so I'm really confused how `nvidia.ko` just doesn't exist where it should?

Snoop05, 2022-02-01:
If you are using dkms its probably in /usr/lib/modules/$(uname -r)/updates/dkms/nvidia.ko.zst (or different extension depending on what compression you have configured)

Prebuilt modules for stock Arch kernel package those to /usr/lib/$(uname)/extramodules/

ThisNekoGuy, 2022-04-11:
It's been some time; apologize for that, my motherboard crapped-out and I needed a new one

This issue would happen, for some reason, often times when re-building `nvidia-all`/`nvidia-tkg` when running the `linux-tkg` kernel; for whatever reason, a very similar error about the same item would pop-up claiming to be missing unless I rebuild the kernel after updating the nvidia driver - which is odd, because that really shouldn't be necessary

Snoop05, 2022-07-09:
Hello again, it appears that you are not building DKMS variant of nvidia-all which is likely the issue requiring you to rebuild it after every kernel update.

simply set `_dkms="true"` in customization.cfg and nvidia rebuilding should be handled automatically

closed 2026-08-19 by damachine