Frogging-Family/nvidia-all #91: Flatpak improvement?

LiamDawe, 2022-02-03:
Hey! So when installing a new driver, Flatpak needs the corresponding runtime too.

This caused a problem when I used nvidia-all to update to NVIDIA 510, as Flatpak didn't have the runtime ready and my OBS Studio broke.

Seems there's a way around this as Robert McQueen (EndlessOS / GNOME Board President) mentioned [on Twitter](https://twitter.com/ramcq/status/1489172949517099018). Is this something you would consider doing to improve nvidia-all for Flatpak use? :)

Bananaman, 2022-02-06:
I dunno what this is about but the Flatpak driver exists now. `sudo flatpak update`

damachine, 2026-08-19:
Thanks but we aren't planning Flatpak and nvidia-all support.

closed 2026-08-19 by damachine