Frogging-Family/nvidia-all #95: 5.18rc patch causes 510.60.02 DKMS to fail on all kernel versions

Daedalusspacegames, 2022-04-06:
When trying to install 510.60.02 using DKMS with kernel versions 5.15.32, 5.16.18, 5.17.1, and 5.18rc1 installed, all DKMS builds fail with more-or-less the same output in the make.log: 
```
DKMS make.log for nvidia-510.60.02 for kernel 5.18.0-rc1-252-tkg-cfs (x86_64)
2022-04-06 09:22:19 -0500
make[1]: Entering directory '/usr/lib/modules/5.18.0-rc1-252-tkg-cfs/build'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 11.2.0
  You are using:           cc (GCC) 11.2.0
  SYMLINK /var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/510.60.02/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_array_uc
 CONFTEST: set_memory_uc
 CONFTEST: acquire_console_sem
 CONFTEST: set_pages_array_uc
 CONFTEST: console_lock
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: acpi_walk_namespace
 CONFTEST: sg_alloc_table
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: efi_enabled
 CONFTEST: pde_data
 CONFTEST: PDE_DATA
 CONFTEST: proc_remove
 CONFTEST: pm_vt_switch_required
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_get_property
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: node_end_pfn
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_remove_bus_device
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: backlight_device_register
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write
 CONFTEST: kthread_create_on_node
 CONFTEST: of_find_matching_node
 CONFTEST: dev_is_pci
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
 CONFTEST: flush_cache_all
 CONFTEST: vmf_insert_pfn
 CONFTEST: jiffies_to_timespec
 CONFTEST: ktime_get_raw_ts64
 CONFTEST: ktime_get_real_ts64
 CONFTEST: full_name_hash
 CONFTEST: hlist_for_each_entry
 CONFTEST: pci_enable_atomic_ops_to_root
 CONFTEST: vga_tryget
 CONFTEST: pgprot_decrypted
 CONFTEST: iterate_fd
[... 160 lines cut ...]
      |          ^
  CC [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-mmap.o
  CC [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-pat.o
  CC [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-procfs.o
  CC [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-procfs-utils.o
  CC [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-usermap.o
  CC [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-vm.o
/var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-dma.c:986: warning: "IMPORT_SGT_STUBS_NEEDED" redefined
  986 | #define IMPORT_SGT_STUBS_NEEDED 0
      | 
/var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-dma.c:980: note: this is the location of the previous definition
  980 | #define IMPORT_SGT_STUBS_NEEDED 1
      | 
make[2]: *** [scripts/Makefile.build:288: /var/lib/dkms/nvidia/510.60.02/build/nvidia/nv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-mmap.c: In function ‘nv_encode_caching’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-mmap.c:343:16: warning: this statement may fall through [-Wimplicit-fallthrough=]
  343 |             if (NV_ALLOW_CACHING(memory_type))
      |                ^
/var/lib/dkms/nvidia/510.60.02/build/nvidia/nv-mmap.c:346:9: note: here
  346 |         default:
      |         ^~~~~~~
make[1]: *** [Makefile:1834: /var/lib/dkms/nvidia/510.60.02/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.18.0-rc1-252-tkg-cfs/build'
make: *** [Makefile:82: modules] Error 2`

Tk-Glitch, 2022-04-06:
Trying to use 5.18 RCs with Nvidia is a terrible idea. Nonethless, fixed with https://github.com/Frogging-Family/nvidia-all/commit/e2514179f35fac4d13a7246984392d0362d7472b

closed 2022-04-06 by Tk-Glitch

ptr1337, 2022-04-06:
Seems to be fixed with non rc kernels now. But for rc it seems still need some stuff patched.
Actual output of dkms:

`  CC [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-format.o
  CC [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia-peermem/nvidia-peermem.o
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-drv.c: In function ‘nv_drm_init_mode_config’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-drv.c:257:21: error: ‘struct drm_mode_config’ has no member named ‘allow_fb_modifiers’
  257 |     dev->mode_config.allow_fb_modifiers = true;
      |                     ^
  LD [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia.o
make[2]: *** [scripts/Makefile.build:288: /var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-drv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘cursor_plane_req_config_update’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c:81:32: warning: unused variable ‘nv_drm_plane_state’ [-Wunused-variable]
   81 |     struct nv_drm_plane_state *nv_drm_plane_state =
      |                                ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c:80:27: warning: unused variable ‘nv_dev’ [-Wunused-variable]
   80 |     struct nv_drm_device *nv_dev = to_nv_device(plane->dev);
      |                           ^~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘plane_req_config_update’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c:182:9: warning: unused variable ‘ret’ [-Wunused-variable]
  182 |     int ret = 0;
      |         ^~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘nv_drm_plane_atomic_set_property’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c:497:32: warning: unused variable ‘nv_drm_plane_state’ [-Wunused-variable]
  497 |     struct nv_drm_plane_state *nv_drm_plane_state =
      |                                ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘nv_drm_enumerate_crtcs_and_planes’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c:1141:13: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 1141 |             struct drm_plane *overlay_plane =
      |             ^~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:71:35: warning: ‘struct dma_buf_map’ declared inside parameter list will not be visible outside of this definition or declaration
   71 |                            struct dma_buf_map *map)
      |                                   ^~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c: In function ‘nv_drm_gem_vmap’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:73:8: error: invalid use of undefined type ‘struct dma_buf_map’
   73 |     map->vaddr = nv_drm_gem_prime_vmap(gem);
      |        ^~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:74:12: error: invalid use of undefined type ‘struct dma_buf_map’
   74 |     if (map->vaddr == NULL) {
      |            ^~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:77:8: error: invalid use of undefined type ‘struct dma_buf_map’
   77 |     map->is_iomem = true;
      |        ^~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c: At top level:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:82:38: warning: ‘struct dma_buf_map’ declared inside parameter list will not be visible outside of this definition or declaration
   82 |                               struct dma_buf_map *map)
      |                                      ^~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c: In function ‘nv_drm_gem_vunmap’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:84:37: error: invalid use of undefined type ‘struct dma_buf_map’
   84 |     nv_drm_gem_prime_vunmap(gem, map->vaddr);
      |                                     ^~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:85:8: error: invalid use of undefined type ‘struct dma_buf_map’
   85 |     map->vaddr = NULL;
      |        ^~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c: At top level:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:98:16: error: initialization of ‘int (*)(struct drm_gem_object *, struct iosys_map *)’ from incompatible pointer type ‘int (*)(struct drm_gem_object *, struct dma_buf_map *)’ [-Werror=incompatible-pointer-types]
   98 |     .vmap    = nv_drm_gem_vmap,
      |                ^~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:98:16: note: (near initialization for ‘nv_drm_gem_funcs.vmap’)
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:99:16: error: initialization of ‘void (*)(struct drm_gem_object *, struct iosys_map *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, struct dma_buf_map *)’ [-Werror=incompatible-pointer-types]
   99 |     .vunmap  = nv_drm_gem_vunmap,
      |                ^~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:99:16: note: (near initialization for ‘nv_drm_gem_funcs.vunmap’)
cc1: some warnings being treated as errors
[... cut ...]
   97 |     bool primary_event = false;
      |          ^~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-modeset.c:96:23: warning: unused variable ‘primary_plane’ [-Wunused-variable]
   96 |     struct drm_plane *primary_plane = crtc->primary;
      |                       ^~~~~~~~~~~~~
make[1]: *** [Makefile:1832: /var/lib/dkms/nvidia/510.60.02/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.18.0-rc1-1-cachyos-rc/build'
make: *** [Makefile:82: modules] Error 2
`