Frogging-Family/nvidia-all #96: 5.18 still fail with the patch

ptr1337, 2022-04-08:
As already commented in https://github.com/Frogging-Family/nvidia-all/issues/95 the issue still exists:

```  CC [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia-peermem/nvidia-peermem.o
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-drv.c: In function ‘nv_drm_init_mode_config’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-drv.c:257:21: error: ‘struct drm_mode_config’ has no member named ‘allow_fb_modifiers’
  257 |     dev->mode_config.allow_fb_modifiers = true;
      |                     ^
  LD [M]  /var/lib/dkms/nvidia/510.60.02/build/nvidia.o
make[2]: *** [scripts/Makefile.build:288: /var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-drv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘cursor_plane_req_config_update’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c:81:32: warning: unused variable ‘nv_drm_plane_state’ [-Wunused-variable]
   81 |     struct nv_drm_plane_state *nv_drm_plane_state =
      |                                ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c:80:27: warning: unused variable ‘nv_dev’ [-Wunused-variable]
   80 |     struct nv_drm_device *nv_dev = to_nv_device(plane->dev);
      |                           ^~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘plane_req_config_update’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c:182:9: warning: unused variable ‘ret’ [-Wunused-variable]
  182 |     int ret = 0;
      |         ^~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘nv_drm_plane_atomic_set_property’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c:497:32: warning: unused variable ‘nv_drm_plane_state’ [-Wunused-variable]
  497 |     struct nv_drm_plane_state *nv_drm_plane_state =
      |                                ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘nv_drm_enumerate_crtcs_and_planes’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-crtc.c:1141:13: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 1141 |             struct drm_plane *overlay_plane =
      |             ^~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:71:35: warning: ‘struct dma_buf_map’ declared inside parameter list will not be visible outside of this definition or declaration
   71 |                            struct dma_buf_map *map)
      |                                   ^~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c: In function ‘nv_drm_gem_vmap’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:73:8: error: invalid use of undefined type ‘struct dma_buf_map’
   73 |     map->vaddr = nv_drm_gem_prime_vmap(gem);
      |        ^~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:74:12: error: invalid use of undefined type ‘struct dma_buf_map’
   74 |     if (map->vaddr == NULL) {
      |            ^~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:77:8: error: invalid use of undefined type ‘struct dma_buf_map’
   77 |     map->is_iomem = true;
      |        ^~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c: At top level:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:82:38: warning: ‘struct dma_buf_map’ declared inside parameter list will not be visible outside of this definition or declaration
   82 |                               struct dma_buf_map *map)
      |                                      ^~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c: In function ‘nv_drm_gem_vunmap’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:84:37: error: invalid use of undefined type ‘struct dma_buf_map’
   84 |     nv_drm_gem_prime_vunmap(gem, map->vaddr);
      |                                     ^~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:85:8: error: invalid use of undefined type ‘struct dma_buf_map’
   85 |     map->vaddr = NULL;
      |        ^~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c: At top level:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:98:16: error: initialization of ‘int (*)(struct drm_gem_object *, struct iosys_map *)’ from incompatible pointer type ‘int (*)(struct drm_gem_object *, struct dma_buf_map *)’ [-Werror=incompatible-pointer-types]
   98 |     .vmap    = nv_drm_gem_vmap,
      |                ^~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:98:16: note: (near initialization for ‘nv_drm_gem_funcs.vmap’)
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:99:16: error: initialization of ‘void (*)(struct drm_gem_object *, struct iosys_map *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, struct dma_buf_map *)’ [-Werror=incompatible-pointer-types]
   99 |     .vunmap  = nv_drm_gem_vunmap,
      |                ^~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.c:99:16: note: (near initialization for ‘nv_drm_gem_funcs.vunmap’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:288: /var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-gem.o] Error 1
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-modeset.c: In function ‘__will_generate_flip_event’:
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-modeset.c:98:10: warning: unused variable ‘overlay_event’ [-Wunused-variable]
   98 |     bool overlay_event = false;
      |          ^~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-modeset.c:97:10: warning: unused variable ‘primary_event’ [-Wunused-variable]
   97 |     bool primary_event = false;
      |          ^~~~~~~~~~~~~
/var/lib/dkms/nvidia/510.60.02/build/nvidia-drm/nvidia-drm-modeset.c:96:23: warning: unused variable ‘primary_plane’ [-Wunused-variable]
   96 |     struct drm_plane *primary_plane = crtc->primary;
      |                       ^~~~~~~~~~~~~
make[1]: *** [Makefile:1832: /var/lib/dkms/nvidia/510.60.02/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.18.0-rc1-1-cachyos-rc/build'
make: *** [Makefile:82: modules] Error 2```

Daedalusspacegames, 2022-04-29:
As of 14718bb2a347c38a1ff896d9032c3726975460e7, even having a 5.18rc kernel installed causes nvidia-all to fail to install:
```
  -> Applying kernel-5.18.patch for 5.18.0-rc4-256-tkg-cfs...
patching file nvidia/nv.c
Hunk #2 FAILED at 2766.
1 out of 2 hunks FAILED -- saving rejects to file nvidia/nv.c.rej
patching file nvidia/linux_nvswitch.c
Hunk #2 FAILED at 2141.
Hunk #3 FAILED at 2166.
Hunk #4 FAILED at 2195.
Hunk #5 FAILED at 2212.
Hunk #6 FAILED at 2235.
Hunk #7 FAILED at 2257.
6 out of 7 hunks FAILED -- saving rejects to file nvidia/linux_nvswitch.c.rej```

ptr1337, 2022-04-29:
I do actually have a patch which works. But the kernel needs also patched. 

Actually through there were some stuff removed from the 5.18 kernel which actually need to get patched. 

The result removing this from dkms is that other kernels are failing in dkms, so just 5.18 will boot.

ptr1337, 2022-04-29:
> As of [14718bb](https://github.com/Frogging-Family/nvidia-all/commit/14718bb2a347c38a1ff896d9032c3726975460e7), even having a 5.18rc kernel installed causes nvidia-all to fail to install:
> 
> ```
>   -> Applying kernel-5.18.patch for 5.18.0-rc4-256-tkg-cfs...
> patching file nvidia/nv.c
> Hunk #2 FAILED at 2766.
> 1 out of 2 hunks FAILED -- saving rejects to file nvidia/nv.c.rej
> patching file nvidia/linux_nvswitch.c
> Hunk #2 FAILED at 2141.
> Hunk #3 FAILED at 2166.
> Hunk #4 FAILED at 2195.
> Hunk #5 FAILED at 2212.
> Hunk #6 FAILED at 2235.
> Hunk #7 FAILED at 2257.
> 6 out of 7 hunks FAILED -- saving rejects to file nvidia/linux_nvswitch.c.rej```
> ```

I’ll watch later to to adjust the patch, seems like they changed some stuff.

Daedalusspacegames, 2022-04-29:
Hacking the PKGBUILD to bypass the patch failure, I was able to sucessfully install the DKMS on 5.18.0-rc4-256-tkg-cfs and boot into is, so 5.18 seems to be fixed as of driver version 510.68.02.

ptr1337, 2022-04-29:
> Hacking the PKGBUILD to bypass the patch failure, I was able to sucessfully install the DKMS on 5.18.0-rc4-256-tkg-cfs and boot into is, so 5.18 seems to be fixed as of driver version 510.68.02.

Wow, cool. Seems hard for nvidia to write this in their patchnotes.. 

I'll close this issue.

closed 2022-04-29 by ptr1337

Daedalusspacegames, 2022-04-29:
Opened a pull request with my workaround: #100 

Does not solve older driver versions on 5.18, but should enable 510.68.02+ to work on kernel 5.18