Frogging-Family/nvidia-all #97: Fails to Build Against Valve's `linux-fsync-headers`

ThisNekoGuy, 2022-04-11:
For some reason I'm not sure of, attempting to install `nvidia-all`/`nvidia-tkg` against Valve's Arch repo kernel fails
(Though probably unrelated, this is also the case for the `xone-dkms-git` kernel module AUR package)

Iglu47, 2022-04-21:
I can't find "Valve's Arch repo kernel" probably deprecated with release 5.16 in favor of the standard Arch kernel.

ThisNekoGuy, 2022-04-21:
> I can't find "Valve's Arch repo kernel" probably deprecated with release 5.16 in favor of the standard Arch kernel.

https://wiki.archlinux.org/title/Unofficial_user_repositories#valveaur

Iglu47, 2022-04-21:
@ThisNekoGuy thanks, I did not know about such a list of repositories, so I really only checked the AUR.

However, this repository appears to be out of date (last updated on 2020-May-28 19:30).
As far as I know, mesa now contains aco and kernel contains fsync, so there should be no reason to use such outdated versions of these programs.
Besides, at the time of 5.6.10 there was probably only the initial fsync `FUTEX_WAIT_MULTIPLE (opcode 31)` which is deprecated today.

Iglu47, 2022-04-21:
For some reason I thought that again there could be a problem with the fact that https://archlinux.org/packages/core/any/linux-api-headers/ has different versions with a custom kernel.
However, there have been similar issues recently with current kernels and nvidia due to a gcc update on Arch (the fix was to just recompile the kernel in the same build environment that tried to build nvidia-dkms).

closed 2026-08-19 by damachine