Frogging-Family/nvidia-all #103: Add support for NVIDIA's open gpu kernel modules

CCF100, 2022-05-12:
https://github.com/NVIDIA/open-gpu-kernel-modules

Tk-Glitch, 2022-05-12:
Added with https://github.com/Frogging-Family/nvidia-all/commit/fa920eb9059907b70c9798e901eccb3dff08e601

closed 2022-05-12 by Tk-Glitch

flindeberg, 2022-05-12:
Sorry for pinging a closed issue, but does anyone have any links to performance tests?

Tk-Glitch, 2022-05-12:
Afaik and outside of missing features, there's exactly zero performance difference.

fidasx, 2022-05-12:
> Afaik and outside of missing features, there's exactly zero performance difference.

here you go
https://www.phoronix.com/scan.php?page=article&item=nvidia-r515-open&num=1

fidasx, 2022-05-12:
> Sorry for pinging a closed issue, but does anyone have any links to performance tests?

https://www.phoronix.com/scan.php?page=article&item=nvidia-r515-open&num=1