Frogging-Family/nvidia-all #104: Nvidia open source drivers

ryanmusante, 2022-05-13:
How will the open source Nvidia drivers be implemented in nvidia-all?

https://www.phoronix.com/scan.php?page=article&item=nvidia-open-kernel&num=1

phush0, 2022-05-13:
It is already implemented, just they don't compile for me.

```
make -C src/nvidia
make -C src/nvidia-modeset
make[1]: Entering directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/src/nvidia'
make[1]: Entering directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/src/nvidia-modeset'
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/src/nvidia-modeset'
cd kernel-open/nvidia-modeset/ && ln -sf ../../src/nvidia-modeset/_out/Linux_x86_64/nv-modeset-kernel.o nv-modeset-kernel.o_binary
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/src/nvidia'
cd kernel-open/nvidia/ && ln -sf ../../src/nvidia/_out/Linux_x86_64/nv-kernel.o nv-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make[2]: Entering directory '/usr/lib/modules/5.17.6-arch1-1/build'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 12.1.0
  You are using:           cc (GCC) 12.1.0
  LD [M]  /tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open/nvidia-modeset.ko
  BTF [M] /tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open/nvidia-modeset.ko
dwarf_expr: unhandled 0x12 DW_OP_ operation
die__process_function: tag not supported 0x2f (template_type_parameter)!
Unsupported DW_TAG_unspecified_type(0x3b)
Encountered error while encoding BTF.
make[4]: *** [scripts/Makefile.modfinal:61: /tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open/nvidia-modeset.ko] Error 1
make[4]: *** Deleting file '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open/nvidia-modeset.ko'
make[3]: *** [scripts/Makefile.modpost:140: __modpost] Error 2
make[2]: *** [Makefile:1746: modules] Error 2
make[2]: Leaving directory '/usr/lib/modules/5.17.6-arch1-1/build'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make: *** [Makefile:50: modules] Error 2
```

fully updated arch 5.17.6-tkg-cfs

tukanoidd, 2022-05-13:
> It is already implemented, just they don't compile for me.

Weird, it built for me no problem, I'm on Garuda tho

phush0, 2022-05-13:
maybe I have removed something from my kernel ???

tukanoidd, 2022-05-13:
> maybe I have removed something from my kernel ???

Don't think so, but looking at the error message it says something about failed encoding. I don't know much about how kernel is written, so can't help you more

phush0, 2022-05-13:
` '!buildflags'` to PKGBUILD fix everything

tukanoidd, 2022-05-13:
> ` '!buildflags'` to PKGBUILD fix everything 

Huh, ok, glad this is fixed for you

Tk-Glitch, 2022-05-13:
> How will the open source Nvidia kernel modules be implemented in nvidia-all?
> 
> https://www.phoronix.com/scan.php?page=article&item=nvidia-open-kernel&num=1
> 
> https://developer.nvidia.com/blog/nvidia-releases-open-source-gpu-kernel-modules/

That way? https://github.com/Frogging-Family/nvidia-all/commit/fa920eb9059907b70c9798e901eccb3dff08e601

In other news, it doesn't seem to compile with Clang (while it works with GCC), and yeah, it's quite sensitive to build flags.

ThisNekoGuy, 2022-05-14:
This was the output of my building with it, with clang:
```
. . .
1 warning generated.
1 warning generated.
date: version.mk: No such file or directory
 [ nvidia            ]  CC           _out/Linux_x86_64/g_nvid_string.c
 [ nvidia            ]  LD           _out/Linux_x86_64/nv-kernel.o
 [ nvidia            ]  OBJCOPY      _out/Linux_x86_64/nv-kernel.o
make[1]: Leaving directory '/home/neko-san/nvidia/nvidia-all/src/open-gpu-kernel-modules-515.43.04/src/nvidia'
cd kernel-open/nvidia/ && ln -sf ../../src/nvidia/_out/Linux_x86_64/nv-kernel.o nv-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/home/neko-san/nvidia/nvidia-all/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make[2]: Entering directory '/home/neko-san/nvidia/nvidia-all/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make[2]: *** /usr/src/linux: No such file or directory.  Stop.
make[2]: Leaving directory '/home/neko-san/nvidia/nvidia-all/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/home/neko-san/nvidia/nvidia-all/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make: *** [Makefile:50: modules] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```
It seemed to build fine until it claimed `/usr/src/linux` didn't exist :/
`!makeflags` made no difference for me

ryanmusante, 2022-05-15:
Does mkinitcpio.conf still need nvidia nvidia_modeset nvidia_uvm nvidia_drm in modules? Do we need modesetting in kernel line? How do we configure open source for PRIME?

closed 2022-05-24 by Tk-Glitch