Frogging-Family/nvidia-all #113: How do I get this package to auto update?

LethalManBoob, 2022-05-26:
Thanks

Tk-Glitch, 2022-06-08:
Unless you feel like scripting your way through it, you don't. It's up to repo maintainers to offer this packageset and update it for their users. Chaotic-aur used to offer it, but I don't know if it's still the case. Regressions and kernel compat breakage between updates made it very annoying for users, so doing manual updates is usually preferred.

LethalManBoob, 2022-06-08:
> Unless you feel like scripting your way through it, you don't. It's up to repo maintainers to offer this packageset and update it for their users. Chaotic-aur used to offer it, but I don't know if [...]

It's no longer in the chaotic aur. Its a shame you cant get it uploaded as a nvidia-all-git as an option but np.