Frogging-Family/nvidia-all #116: Rename drivers

TechXero, 2022-06-06:
Would it be possible to rename resulting packages so we can host different versions on a single repo ?

If so how ?

Tk-Glitch, 2022-06-06:
The resulting packages names are already versioned, so I'm not sure I understand your request.

TechXero, 2022-06-06:
I mean all resulting packages are named nvidia-dkms-tk-version but on a repo only name is taken, if one version is higher it will just include highest version number ignoring older so can't add 470 and 510 and 515 on same repo only 515 will be added.. I hope this explains it..

Tk-Glitch, 2022-06-06:
Oh I see now. That would take adding some fluff to allow for a variable in pkgname.

TechXero, 2022-06-06:
Kind of.. Best way is for 470 for example package and related deps to be called nvidia-470-dkms etc.. Y'know ?

TechXero, 2022-06-06:
Sure.. 

Jonathon on AUR does it best but yours include TKG patches and for some reason perform better

[image]

TechXero, 2022-06-07:
Sorry to bother on closed issue but I built with changes still not as image I posted above won't allow me to include more than one version on repo

Tk-Glitch, 2022-06-08:
Did you enable the new option in your .cfg?

TechXero, 2022-06-08:
What do you mean ? Can you be more specific ?

Tk-Glitch, 2022-06-08:
You need to set the newly added option to "true": https://github.com/Frogging-Family/nvidia-all/blob/90e0b9e0b67c0208e3ca153be5941a2c3634c2c5/customization.cfg#L14

TechXero, 2022-06-08:
Done thanks ... Now I have 510 & 470 once tests are done so will 515 ;)

closed 2022-06-08 by TechXero