Frogging-Family/nvidia-all #117: Difference in package in arch repo (egl-gbm lib)

czapnaper, 2022-06-07:
In arch repo [nvidia-utils](https://archlinux.org/packages/extra/x86_64/nvidia-utils/files/) package  conteins `/usr/lib/libnvidia-egl-gbm.so/usr/lib/libnvidia-egl-gbm.so.1 /usr/lib/libnvidia-egl-gbm.so.1.1.0 /usr/share/egl/egl_external_platform.d/15_nvidia_gbm.json` in this repo nvidia-egl-wayland-tkg package conteins this files.

Tk-Glitch, 2022-06-08:
We have other differences vs Arch's nvidia packages, but is that an issue? Having egl libs in the egl package makes more sense to me, but if there's an actual issue with that please share. Thanks!

czapnaper, 2022-06-08:
This is a problem when  use egl-wayland or  egl-wayland-git package (don't contain this files). I noticed this when I wanted to test the latest version of egl-wayland ([1.1.10](https://github.com/NVIDIA/egl-wayland/commit/247335d5d451bc56af08781c70f8bc96926cbf23)).

closed 2022-06-09 by Tk-Glitch