Frogging-Family/nvidia-all #120: Black Screen

TechXero, 2022-07-25:
Hey...

After rebuilding 515-dkms drivers at r214 and installing getting black screen with AMD Kernel.. I dunno why ! On mainline Arch Kernel no issues booting to OS ..

Tk-Glitch, 2022-07-26:
What's "AMD Kernel"? Anyway if it works with Arch kernel it's not an issue with nvidia-all. Nvidia modules can't be compiled against any kernel, and if interfaces used by them were modified in a way that's no handled, it will fail.

TechXero, 2022-07-26:
It used to work, was working fine on r212 when I updated to r214 it broke.. Also I use dkms .. And AMD kernel is a Ryzen optimized Kernel.. Something happened between r212 n r214 that caused it to break ?

Tk-Glitch, 2022-07-26:
First, nvidia-all aren't custom drivers. You get what Nvidia provides.
Secondly, 212 introduced 515.57, 213 introduced 515.49.06 (vulkan dev) and 214 introduced 515.49.10 (vulkan dev). There was exactly zero change to anything we handle between 212 and 214. If you're still using the same driver revision and the same choice regarding open-source kernel modules, the package is effectively exactly the same.

Tk-Glitch, 2022-07-26:
I've been using 515.57 and 515.49.10 fine on linux-tkg 5.18, so it's something else that broke it for you.

TechXero, 2022-07-26:
I use 515.57 not 515.49.10 I did nothing special.. All I did was update to r214 nothing more

Tk-Glitch, 2022-07-26:
Also, there are *many* nvidia-all + custom kernel users out there, and you're the first to report such an issue. Nobody pinged me on Discord either.

Tk-Glitch, 2022-07-26:
> I use 515.57 not 515.49.10 I did nothing special.. All I did was update to r214 nothing more

r214 didn't change anything for 515.57, so either your kernel is broken or something turned bad in the chain somewhere. But no change in nvidia-all for sure.

TechXero, 2022-07-26:
Like I said all I did was build 515.57 r214 and update nothing more now I get black screen.. And talked to AMD Kernel maintainer he said he did nothing special.. Could it be missing patches ?

Tk-Glitch, 2022-07-26:
If we changed anything in there I'd gladly track the bug out, but in this case there's simply no change :/

TechXero, 2022-07-26:
Strange.. Nature of Arch I guess.. Will wait for next AMD Kernel update n use mainline one in meantime.. No biggie, keep this open will reply with any changes ..

Thanks ;)

Tk-Glitch, 2022-07-26:
Also, if you don't believe me and wanted to go back to r212, you can `git checkout aaf1bc0dbc91a4b29f9f096a3936363f9930f499` in your nvidia-all clone.

Tk-Glitch, 2022-07-26:
Could you link that "AMD Kernel" source if possible? The fact that it works with Arch's (and linux-tkg custom kernel, as tested by many) tends to point to an issue with that specific kernel.

TechXero, 2022-07-26:
Nahh it's ok will wait for Kernel update, if nothing works I will have to roll back.. Problem is I host them on my repo for my users.. In case anyone reports issue will let you know...

AMD Kernel source (AUR)
https://aur.archlinux.org/packages/linux-amd

Tk-Glitch, 2022-07-26:
Thanks. Indeed it doesn't seem to do much. Graysky's patch and.. That's about it? If it's otherwise a plain copy of Arch's kernel there should be no difference in behavior (building with `-march=whatever` doesn't do much). That's definitely weird.

TechXero, 2022-07-26:
My feeling exactly.. I am not a dev at all just a regular user who just knows how to build a custom Distro.. So I do not understand code xD

ptr1337, 2022-07-26:
I think just the kernel needs a rebuild.
GCC has been updated the last 3 days to 12.0.3 which maybe results into that your dkms is failing. 
Just reinstall the kernel and check if your dkms that output any error - if yes rebuild the kernel

TechXero, 2022-07-26:
> I think just the kernel needs a rebuild. GCC has been updated the last 3 days to 12.0.3 which maybe results into that your dkms is failing. Just reinstall the kernel and check if your dkms that output any error - if yes rebuild the kernel

I thought about that I re-installed Kernel nothing.. No errors and still black screen

Tk-Glitch, 2022-07-26:
Since it enforces `znver3` are you using a Zen3 CPU?

TechXero, 2022-07-26:
Ryzen 5 3600x so yeah....

Tk-Glitch, 2022-07-26:
That's not zen3, that's zen2. For desktop CPUs, Zen3 are 5000 series.
With the retbleed mitigations in place with 5.18.14, it's not unlikely that the kernel is now more sensitive to zen3 optimizations on your zen2 CPU, breaking things. What you could try to validate that theory is to change that line in your custom kernel PKGBUILD: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=linux-amd#n81 so it's using `-march=znver2 -mtune=znver2` instead of the current `-march=znver3 -mtune=znver3`.
Nonetheless `znver3` is totally incorrect for your CPU.

TechXero, 2022-07-26:
Ok so it's Zen 2 damn so it's now enforced. Was working before... Flawless... I guess will have to rebuild...

Good catch thanks...

Tk-Glitch, 2022-07-26:
Considering that custom kernel's seemingly only changes are related to those µarch optimizations, that's the most likely culprit. I'll be crossing my fingers for you 😄

TechXero, 2022-07-26:
Will let you know. No power right now...

TechXero, 2022-07-26:
Just tried znver2 for Zen2 same issue.. I am now compiling tkg Kernel with zen2 patches to see if that works.. If it fails I dunno what to do except stick with mainline Arch Kernel xD

LethalManBoob, 2022-09-21:
latest driver gives me black screen. had to chroot in an old version.

Tk-Glitch, 2022-09-21:
Booted just fine on my end with a 980ti on Z370. Anyway this issue isn't related and I'll close it as the root cause was found :frog:

closed 2022-09-21 by Tk-Glitch