Frogging-Family/nvidia-all #121: Nothing but v.470 is working on my end

mozo78, 2022-07-27:
Hello,
I'm using nvidia-all for years without any problems at all but now I have a problem. It seems I have a blacj screen with any newer version than 470. If I install a newer version, after the reboot it stucks on "starting version 251..." and that's it. It waits forever but nothing happens. The driver installation is working fine and gives no errors. I tried with the open source module and without it to no avail. Now I'm stuck with 470 - the only version which allows me to reach my desktop graphically. I have a backup and I'm not worried at all but I'm wondering what could be the problem?
Thank you for your hard work and great projects!

Tk-Glitch, 2022-07-27:
Would you happen to have an Alderlake CPU?

Tk-Glitch, 2022-07-27:
Also, on which GPU?

mozo78, 2022-07-27:
I'm with 9900K and GTX 1080Ti :)

Tk-Glitch, 2022-07-27:
Ok so that shouldn't require anything fancy.. Weird. No failure in dkms? Using a custom kernel?

mozo78, 2022-07-27:
Nope, It's even not the LTS. It's the default kernel.

mozo78, 2022-07-27:
Yep, X11. I never tried Wayland because of NVIDIA.

Tk-Glitch, 2022-07-27:
Do you have a `/etc/X11/xorg.conf` and/or some nvidia.conf in `/etc/X11/xorg.conf.d/`, it might be worth deleting it/them.

mozo78, 2022-07-27:
Yes, I have /etc/X11/xorg.conf:
https://pastebin.com/B9z59vua

I'm using this config for years with no problems. I have a dual boot with Mint and it's working fine there too but I'm not using it. It's just for game testing :)

mozo78, 2022-07-30:
I found it:
https://bugs.archlinux.org/task/74886
https://bugs.archlinux.org/task/74891

All is working fine now :)

closed 2022-07-30 by mozo78