Frogging-Family/nvidia-all #126: open source drivers - ERROR: A failure occurred in build()

ryanmusante, 2022-10-07:
Unable to build open source drivers for 515.49.19 & 515.76

Error messages cause build to fail at end. What am I doing wrong?
```
 [ nvidia-modeset    ]  LD           _out/Linux_x86_64/nv-modeset-kernel.o
make[1]: Leaving directory '/home/ryan/nvidia-all/src/open-gpu-kernel-modules-515.49.19/src/nvidia-modeset'
cd kernel-open/nvidia-modeset/ && ln -sf ../../src/nvidia-modeset/_out/Linux_x86_64/nv-modeset-kernel.o nv-modeset-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/home/ryan/nvidia-all/src/open-gpu-kernel-modules-515.49.19/kernel-open'
make[2]: Entering directory '/home/ryan/nvidia-all/src/open-gpu-kernel-modules-515.49.19/kernel-open'
make[2]: *** /usr/src/linux: No such file or directory.  Stop.
make[2]: Leaving directory '/home/ryan/nvidia-all/src/open-gpu-kernel-modules-515.49.19/kernel-open'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/home/ryan/nvidia-all/src/open-gpu-kernel-modules-515.49.19/kernel-open'
make: *** [Makefile:50: modules] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
  -> exit cleanup done
```

```
 Kernel: 6.0.0-270-tkg-pds arch: x86_64 bits: 64 compiler: gcc v: 12.2.0
```

Tk-Glitch, 2022-10-08:
`/usr/src/linux: No such file or directory`
Missing headers?

ryanmusante, 2022-10-09:
I have headers for both kernels installed [default & tkg]
```
~/nvidia-all master > pacman -Qi | grep -i headers
Description     : Access control list utilities, libraries and headers
Description     : Free peer-reviewed portable C++ source libraries (development headers)
                  llvm: referenced by some clang headers [installed]
Optional Deps   : linux-headers: build modules against the Arch kernel [installed]
                  linux-lts-headers: build modules against the LTS kernel
                  linux-zen-headers: build modules against the ZEN kernel
                  linux-hardened-headers: build modules against the HARDENED kernel
Required By     : autoconf  base  bashtop  dkms  linux60-headers  mkinitcpio  pacman
Depends On      : linux-api-headers>=4.10  tzdata  filesystem
Optional Deps   : opencl-headers: headers necessary for OpenCL development [installed]
Depends On      : glibc  libelf  linux-api-headers
Description     : The CUPS Printing System - client libraries and headers
Required By     : binutils  gstreamer  iproute2  lib32-libelf  libbpf  libva-mesa-driver  linux60-headers  mesa  mesa-vdpau  pahole  systemd  vulkan-radeon
Name            : linux-api-headers
Description     : Kernel headers sanitized for use in userspace
Name            : linux60-headers
Provides        : linux-headers=6.0.0
Name            : linux60-tkg-pds-headers
Description     : Headers and scripts for building modules for the Linux-tkg kernel - https://github.com/Frogging-Family/linux-tkg
Provides        : linux-headers=6.0.0  linux60-tkg-pds-headers=6.0.0
Optional Deps   : linux-headers [installed]
                  linux-lts-headers: Build the module for LTS Arch kernel
Optional Deps   : linux-headers [installed]
                  linux-lts-headers: Build the module for LTS Arch kernel
Optional Deps   : linux-headers [installed]
                  linux-lts-headers: Build the module for LTS Arch kernel
Name            : opencl-headers
Optional Deps   : opencl-headers: headers necessary for OpenCL development [installed]
Required By     : linux60-headers  linux60-tkg-pds-headers  nvidia-dkms-tkg
Required By     : apparmor  flatpak  fwupd  geany-plugins  gnome-todo  gst-editing-services  gufw  kitty  kitty-shell-integration  lightdm-slick-greeter  linux60-headers  manjaro-application-utility  menulibre  pacman-mirrors  pahole  pastebinit  pkgbrowser  proton-ge-custom-bin  python-appdirs  python-autocommand  python-bcrypt  python-cachetools  python-cairo  python-certifi  python-chardet  python-click  python-distro  python-evdev  python-filelock  python-future  python-gobject  python-httplib2  python-idna  python-inflect  python-jaraco.context  python-lxml  python-manjaro-sdk  python-more-itertools  python-npyscreen  python-ordered-set  python-pillow  python-pip  python-ply  python-protobuf  python-psutil  python-pulsectl  python-pyasn1  python-pydantic  python-pydrive2  python-pygments  python-pynacl  python-pyparsing  python-pyqt5-sip  python-pytz  python-pyxdg  python-reportlab  python-setproctitle  python-six  python-soupsieve  python-systemd  python-tomli  python-trove-classifiers  python-typing_extensions  python-uritemplate  python-urllib3  python-xapp  python-yaml  python-zipp  samba  smbclient  steam  ufw
                  perl: for fixqt4headers and syncqt [installed]
Depends On      : libjpeg-turbo  xcb-util-keysyms  xcb-util-cursor  libgl  fontconfig  xdg-utils  shared-mime-info  xcb-util-wm  libxrender  libxi  sqlite  mesa  vulkan-headers  tslib  libinput  libxkbcommon-x11  libproxy  libcups  double-conversion  brotli  libb2  md4c  qt6-translations
Optional For    : linux60-tkg-pds-headers
Name            : vulkan-headers
Description     : combined X.Org X11 Protocol headers
```

Tk-Glitch, 2022-10-10:
Check if `/usr/src/linux` exists

ryanmusante, 2022-10-10:
I made the directory, but now get hit with "No rule to make target modules
```
 [ nvidia-modeset    ]  CXX          src/dp/nvdp-timer.cpp
date: version.mk: No such file or directory
 [ nvidia-modeset    ]  CC           _out/Linux_x86_64/g_nvid_string.c
 [ nvidia-modeset    ]  LD           _out/Linux_x86_64/nv-modeset-kernel.o
make[1]: Leaving directory '/home/ryan/nvidia-all/src/open-gpu-kernel-modules-515.76/src/nvidia-modeset'
cd kernel-open/nvidia-modeset/ && ln -sf ../../src/nvidia-modeset/_out/Linux_x86_64/nv-modeset-kernel.o nv-modeset-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/home/ryan/nvidia-all/src/open-gpu-kernel-modules-515.76/kernel-open'
make[2]: Entering directory '/usr/src/linux'
make[2]: *** No rule to make target 'modules'.  Stop.
make[2]: Leaving directory '/usr/src/linux'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/home/ryan/nvidia-all/src/open-gpu-kernel-modules-515.76/kernel-open'
make: *** [Makefile:50: modules] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
  -> exit cleanup done
~/nvidia-all master >                    
```

Valmar33, 2022-11-11:
> `linux-tkg-headers` appears to make the directory `/usr/src/linux60-tkg-{scheduler}` rather than placing the headers in `/usr/src/linux`.

This is supposed to happen, otherwise linux-tkg-headers would have file conflicts with linux-tkg-headers.

/usr/src/<kernel> is just a symlink to /usr/lib/modules/<kernel version>/build anyways.

RA3236, 2022-11-11:
> This is supposed to happen, otherwise linux-tkg-headers would have file conflicts with linux-headers.
>/usr/src/>kernel< is just a symlink to /usr/lib/modules/>kernel version</build anyways

Right, that makes more sense. I was wondering why rebooting preventing the graphical session from starting :)

Valmar33, 2022-11-11:
> I had the same issue, but moving the directory contents to the correct location fixed it.

This is just a hacky cludge. It's not the "correct" location so much as the open source Nvidia module seemingly being hardcoded to expect /usr/src/linux to be the location to find its desired headers.

Tk-Glitch, 2022-11-29:
The symlink is always correctly created when installing either stock linux packages or linux-tkg on my end, on Arch and EndeavourOS. Are you guys hitting the issue using a different distro?

ryanmusante, 2022-11-29:
I'm using Manjaro Linux

closed 2023-06-22 by ryanmusante