Frogging-Family/nvidia-all #128: Question regaeding Fedora Kernel 6.0.5 and and Nvidia 470 drivers

vvstubbs, 2022-10-31:
As I only learned about this project yesterday, I'm just wanting to clarify as what state I should have my now broken X windows configuration since the upgrade to the newly released fedora 6.0.5 on the XFCE spin with the installed GeForce GT 730. Currently none of the options work,:
1) the real Nvidia 470.141.03 that was working with 5.19.16-200.fc36.x86_64
2) the "sudo dnf install xorg-x11-drv-nvidia-470xx akmod-nvidia-470xx xorg-x11-drv-nvidia-470xx-cuda"
3) or the Nouveau drivers broken due to missing or incomplete Xorg.conf
So currently the only method that allows me to get X working is to use GRUB to load the 5.19.16-200 kernel instead of the 6.0.05 kernel and install the Nvidia 470.141.03 drivers, Certainly not an ideal situation and quiet honestly a dogs breakfast. So before starting down the path of using this project, I would like to know at what state should I have the machine before beginning ? Any comments or suggestions would be most appreciated.

ptr1337, 2022-10-31:
The current 470 and 390 drivers are broken with the 6.0 Kernel.
 There is a patch for the 470 drivers which should work.
 
 I did test a 390xx patch for 6.0, dkms compiled fine but xorg crashes so far.
 
 Here is the patch for 470xx:
 https://aur.archlinux.org/cgit/aur.git/plain/kernel-6.0.patch?h=nvidia-470xx-utils

vvstubbs, 2022-11-02:
Thanks for the information. I decided to just go ahead and upgrade my video card to something the is supported by the currently available drivers, namely 515.76. So I decided on a GT 1030, relatively cheap and quickly available

closed 2022-11-02 by vvstubbs