Frogging-Family/nvidia-all #131: cannot stat 'firmware/gsp.bin'

BlueGoliath, 2022-11-10:
When attempting to manually add and install the 525.53 release by editing the customization config I get an error:

`Starting package_nvidia-utils-tkg()...
install: cannot stat 'firmware/gsp.bin': No such file or directory
`

This is under a GTX 1080 which does not support GSP.

qumaciel, 2022-11-10:
The 525.53 driver has multiple GSP firmwares, cf. [PR#130](https://github.com/Frogging-Family/nvidia-all/pull/130).

qumaciel, 2022-11-10:
The module decides to load the firmware at runtime, in case your card doesn't need it, it will not load it.

So, there is no need for not having the firmware(s) installed at the proper location. 
In case you hotplug an eGPU or decide to change your card, the firmwares will be there to load and the driver will work as intended.

closed 2022-12-15 by Tk-Glitch