Frogging-Family/nvidia-all #146: ERROR: Failure while downloading

pirolisi, 2023-02-04:
Hi I am having some trouble updating the vulkan drivers (I am currently on 515.49.25 vulkan), the error is this:
curl: (22) The requested URL returned error: 404
==> ERROR: Failure while downloading https://developer.nvidia.com/vulkan-beta-5254706-linux

tried also with the new version 525.47.07 but the same error shows up, it is a broken link or am I doing something wrong?
I have noticed that the actual link (for the newest vulkan driver at least) on nvidia page is 
"https://developer.nvidia.com/downloads/vulkan-beta-5254707-linux"
instead of the one actually showing up executing the script 
"https://developer.nvidia.com/vulkan-beta-5254707-linux"

these are the steps I follow (Manjaro Gnome): 
  What driver version do you want?
    > 1.Vulkan dev: 515.49.14
      2.515 series: 515.65.01
      3.510 series: 510.73.05
      4.495 series: 495.46
      5.470 series: 470.141.03
      6.Older series
      7.Custom version (396.xx series or higher)
    choice[1-7?]: 7
What branch do you want?
> 1.Stable or regular beta
  2.Vulkan dev
choice[1-2?]: 2
Type the desired version number (examples: 415.18.02, 396.54.09): 525.47.06
Build the dkms package or the regular one?
> 1.dkms (recommended)
  2.regular
choice[1-2?]: 1
  ->  - Open source kernel modules available - 
==> WARNING: IT ONLY OFFERS SUPPORT FOR TURING AND NEWER, AND DOESN'T OFFER ALL THE FEATURES OF THE PROPRIETARY ONE.
==> WARNING: PRIME SUPPORT, VRR AND POWER MANAGEMENT ARE NOTABLY MISSING CURRENTLY.
    Do you want to use it instead of the proprietary one?
    > N/y : n
  -> Building driver version 525.47.06 on branch vulkandev.
  -> Selected driver integrity check behavior (md5sum or SKIP): SKIP
==> Making package: nvidia-dev-dkms-tkg 525.47.06-218 (Sat 04 Feb 2023 09:39:38 EET)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Downloading NVIDIA-Linux-x86_64-525.47.06.run...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 61363    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
==> ERROR: Failure while downloading https://developer.nvidia.com/vulkan-beta-5254706-linux
    Aborting...
  -> exit cleanup done

Thanks in advance for any help.

Tk-Glitch, 2023-02-04:
You are using an old version of nvidia-all which doesn't even support 525 series. Please update your tree with `git pull` or by doing a fresh clone with `git clone https://github.com/Frogging-Family/nvidia-all.git`.

pirolisi, 2023-02-05:
> You are using an old version of nvidia-all which doesn't even support 525 series. Please update your tree with `git pull` or by doing a fresh clone with `git clone https://github.com/Frogging-Family/nvidia-all.git`.

Thank you, resolved

closed 2023-07-08 by Tk-Glitch