Frogging-Family/nvidia-all #153: dkms error on linux 66.3rc1

pipes80-collab, 2023-03-07:
good evening, I would like to thank you for the good work you are doing with this nvidia installation package.
i have a problem with linux kernel 6.3 rc1
==> WARNING: `dkms install --no-depmod nvidia/530.30.02 -k 6.3.0-rc1' exited 10

ptr1337, 2023-03-08:
Hi,

actually the driver needs to be patched. I already create an issue at the GitHub. 

There is a working patch for the 525 series

pipes80-collab, 2023-03-09:
Think you that patch work on 530?

ptr1337, 2023-03-10:
https://gist.github.com/joanbm/d10e9cbbbb8e245b6e7e27b2db338faf

Here the patch for 525. 
Upstream issue:
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/469

pipes80-collab, 2023-03-10:
this is the make.log
`DKMS make.log for nvidia-530.30.02 for kernel 6.3.0-rc1-x64v3 (x86_64)
ven 10 mar 2023, 20:43:23, CET
make[1]: ingresso nella directory «/home/filippo/Scaricati/linux-6.3-rc1»
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 12.2.1 20230201
  You are using:           cc (GCC) 12.2.1 20230201
  SYMLINK /var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/530.30.02/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: set_pages_array_uc
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: ioremap_wc
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_rebar_get_possible_sizes
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write_has_pointer_pos_arg
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
 CONFTEST: flush_cache_all
 CONFTEST: vmf_insert_pfn
 CONFTEST: jiffies_to_timespec
 CONFTEST: ktime_get_raw_ts64
 CONFTEST: full_name_hash
 CONFTEST: ktime_get_real_ts64
 CONFTEST: pci_enable_atomic_ops_to_root
 CONFTEST: vga_tryget
 CONFTEST: pgprot_decrypted
 CONFTEST: cc_mkdec
 CONFTEST: seq_read_iter
 CONFTEST: unsafe_follow_pfn
 CONFTEST: drm_gem_object_get
 CONFTEST: drm_gem_object_put_unlocked
 CONFTEST: add_memory_driver_managed
 CONFTEST: device_property_read_u64
 CONFTEST: devm_of_platform_populate
 CONFTEST: of_dma_configure
 CONFTEST: of_property_count_elems_of_size
 CONFTEST: of_property_read_variable_u8_array
 CONFTEST: of_property_read_variable_u32_array
 CONFTEST: i2c_unregister_device
 CONFTEST: i2c_new_client_device
 CONFTEST: of_get_named_gpio
 CONFTEST: devm_gpio_request_one
 CONFTEST: gpio_direction_input
[... 204 lines cut ...]
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c: In function ‘nvidia_mmap_helper’:
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:606:23: error: assignment of read-only member ‘vm_flags’
  606 |         vma->vm_flags |= VM_IO | VM_PFNMAP | VM_DONTEXPAND;
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:673:23: error: assignment of read-only member ‘vm_flags’
  673 |         vma->vm_flags |= (VM_IO | VM_LOCKED | VM_RESERVED);
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:674:23: error: assignment of read-only member ‘vm_flags’
  674 |         vma->vm_flags |= (VM_DONTEXPAND | VM_DONTDUMP);
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:680:23: error: assignment of read-only member ‘vm_flags’
  680 |         vma->vm_flags &= ~VM_WRITE;
      |                       ^~
/var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.c:681:23: error: assignment of read-only member ‘vm_flags’
  681 |         vma->vm_flags &= ~VM_MAYWRITE;
      |                       ^~
  CC [M]  /var/lib/dkms/nvidia/530.30.02/build/nvidia/linux_nvswitch.o
  CC [M]  /var/lib/dkms/nvidia/530.30.02/build/nvidia/procfs_nvswitch.o
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/nvidia/530.30.02/build/nvidia/nv-mmap.o] Error 1
make[2]: *** Attesa per i processi non terminati....
  CC [M]  /var/lib/dkms/nvidia/530.30.02/build/nvidia/i2c_nvswitch.o
make[1]: *** [Makefile:2028: /var/lib/dkms/nvidia/530.30.02/build] Error 2
make[1]: uscita dalla directory «/home/filippo/Scaricati/linux-6.3-rc1»
make: *** [Makefile:82: modules] Error 2
`

ptr1337, 2023-03-23:
Fixed in 530.41.03

pipes80-collab, 2023-03-23:
solved, thanks for good work