Frogging-Family/nvidia-all #166: install fails (nvidia-tkg.hook ... FAILED)

elogicuk, 2023-05-05:
here is some info about my system.

OS: Manjaro Linux x86_64 
Host: MS-7B17 2.0 
Kernel: 5.15.108-1-MANJARO 
DE: GNOME 43.4 
GPU: NVIDIA GeForce RTX 2060 SUPER 

----------------

==> Validating source files with md5sums...
    NVIDIA-Linux-x86_64-525.47.13.run ... Passed
    10-nvidia-drm-outputclass.conf ... Passed
    nvidia-utils-tkg.sysusers ... Passed
    60-nvidia.rules ... Passed
    **nvidia-tkg.hook ... FAILED**
    nvidia-open-gcc-ibt-sls.diff ... Passed
    linux-version.diff ... Passed
    01-ipmi-vm.diff ... Passed
    02-ipmi-vm.diff ... Passed
    list_is_first.diff ... Passed
    kernel-4.16.patch ... Passed
    kernel-4.19.patch ... Passed
    kernel-5.0.patch ... Passed
    kernel-5.1.patch ... Passed
    kernel-5.2.patch ... Passed
    kernel-5.3.patch ... Passed
    kernel-5.4.patch ... Passed
    kernel-5.4-symver.diff ... Passed
    kernel-5.4-prime.diff ... Passed
    kernel-5.5.patch ... Passed
    kernel-5.6.patch ... Passed
    5.6-legacy-includes.diff ... Passed
    5.6-ioremap.diff ... Passed
    kernel-5.7.patch ... Passed
    kernel-5.8.patch ... Passed
    5.8-legacy.diff ... Passed
    kernel-5.9.patch ... Passed
    5.9-gpl.diff ... Passed
    kernel-5.10.patch ... Passed
    kernel-5.11.patch ... Passed
    5.11-legacy.diff ... Passed
    455-crashfix.diff ... Passed
    kernel-5.12.patch ... Passed
    kernel-5.14.patch ... Passed
    kernel-5.16.patch ... Passed
    kernel-5.16-std.diff ... Passed
    kernel-5.17.patch ... Passed
    kernel-6.0.patch ... Passed
    kernel-6.0-470.patch ... Passed
    kernel-6.2.patch ... Passed
==> ERROR: One or more files did not pass the validity check!
  -> exit cleanup done

Tricked-dev, 2023-05-05:
i fixed it by changing line 261 in the `PKGBUILD` to 
```
         '596f7cbf2db48d4f5b1c38967bb93cea'
```

elogicuk, 2023-05-05:
> i fixed it by changing line 261 in the `PKGBUILD` to
> 
> ```
>          '596f7cbf2db48d4f5b1c38967bb93cea'
> ```

hmm didnt work for me, caused even more issues.

  -> Building driver version 525.47.13 on branch vulkandev.
==> ERROR: A (correctly made?) linux-headers package can't be found.
    If you're sure it's installed, blame your kernel maintainer.
    Press enter to proceed anyway...
/home/home/Videos/nvidia-all/PKGBUILD: line 261: 596f7cbf2db48d4f5b1c38967bb93cea: command not found
  -> Selected driver integrity check behavior (md5sum or SKIP): 5b044d7f31637ff5c8f6459f10da727d
==> ERROR: Missing package_nvidia-dev-tkg() function for split package 'nvidia-dev-tkg'
  -> exit cleanup done

Tricked-dev, 2023-05-05:
> > i fixed it by changing line 261 in the `PKGBUILD` to
> > ```
> >          '596f7cbf2db48d4f5b1c38967bb93cea'
> > ```
> 
> i think you mean line 361?

oh yeah i did, i edited my reply

rotteegher39, 2023-05-05:
Having the same issue on Archlinux
OS: Arch Linux x86_64
Kernel: 6.3.1-arch1-1
WM: LeftWM on X11
GPU: NVIDIA GeForce RTX 2060 mobile

Changing the 361 line as you mentioned worked fine.

elogicuk, 2023-05-05:
i installed these drivers and it totally broke my system 

had to bring my system back to life with this command

 sudo mhwd -a pci nonfree 0300

gardotd426, 2023-05-07:
You can close this issue, the drivers have been updated and the vulkan dev branch works fine.

ryanmusante, 2023-06-27:
@elogicuk do you have remnants of other Manjaro driver install? Just curious

closed 2023-07-08 by Tk-Glitch