Frogging-Family/nvidia-all #177: Nvidia 530.41.03 won't build with kernel 6.4

ryanmusante, 2023-06-27:
Related to post below, nv 530.41.03 will not build with kernel 6.4. Others having same problem? 

https://forums.developer.nvidia.com/t/error-compiling-with-recent-mainline-kernel-linux-6-4-rc1/252732

thojohns, 2023-07-02:
The linked thread already contains a solution. [Granted, it's a solution for 470.xx](https://gist.github.com/joanbm/77f0650d45747b9a4dc8e330ade2bf5c), but the code that needs changing is the same - it's only the line numbers that differ, although you'd also want to modify files in /kernel-dkms instead of /kernel.

[I threw together a quick patch for 530.41.03](https://gist.github.com/thojohns/61f27ce33ef8f5d390936d27051ba36b) based on the 470 patch, which you'll need to put in your patches folder. [I also updated the PKGBUILD to use it](https://gist.github.com/thojohns/608332ff7f1c2b913364d81aed81efc2). 

I'm refraining from submitting a pull request with this fix because A) it's not a general solution - every driver version pre-535 will probably need a patch like this unless Nvidia puts out minor releases with a similar fix, and B) I've been having a hell of a time trying to update the PKGBUILD with a section for 6.4 patches - I initially added a tweaked 470 patch as well but no matter what conditionals I set, the script kept trying to use both the 470 and 530 patches, resulting in a build failure.

ryanmusante, 2023-07-02:
Ahh, nice. When I get the chance I will test it

Drogobo, 2023-07-05:
this did fix the compile errors.

thojohns, 2023-07-06:
I believe Nvidia has already implemented their own similar fix - obviously, 535 compiles without issue, and if you look at the open kernel modules you'll even see that their fix is slightly more elegant despite accomplishing the exact same result, but there's also a workaround included in the 470.199 driver, so it does seem like it's being backported. I don't expect any future releases for 470, 535, or otherwise will suffer from this issue in particular.

closed 2023-07-06 by Tk-Glitch