Frogging-Family/nvidia-all #184: [Question to TK] What packages would I need to rollback to kernel 5.15 and nvidia-dkms 515.57?

rado84-github, 2023-08-18:
The latest drivers (by that I mean versions from 525.XX.XX) are full of crap and underthought things, so I wanna rollback my system to kernel 5.15 LTS (I have the packages for that), so I installed the system anew, installed kernel 5.15 and booted into it - all went flawlessly.
But when I installed nvidia-dkms 515.57, the system wouldn't boot and hung at either "Reached Graphical Target" or "Started Account Services" and the desktop never comes online. I downloaded all the packages that I know of but apparently there are other packages that I've missed. I've downloaded and stored all the files and dependencies that appeared somewhere as such. Thus so far I've got ccache (that's not a dep, but a conveniency), dkms, gcc, gcc-libs, lib32-gcc-libs. As for the driver itself I got these packages, all version 515.57: nvidia-utils, lib32-nvidia-utils, nvidia-utils, nvidia-dkms, nvidia-settings. These 3 also appeared as a dep somewhere, so I downloaded them too: libglvnd, libxext, mesa. But even with these new 3 the system still won't boot to desktop, so what else am I missing here?

The drivers for my videocard end with 515.76, everything else after that is related to the RTX cards which I don't have and I probably won't have for another 5 years, so I really wanna go back to the good drivers versions where everything simply worked. But for that to happen I need to know what other packages I must download from Arch's archive.

Tk-Glitch, 2023-08-18:
What's your GPU? I'm not seeing an official statement regarding dropping support for legacy series with 515.76, unless you meant most changes on newer released wouldn't affect your card.

Shooting in the dark here, but GCC packages were updated recently so if your 5.15 kernel wasn't built against them the Nvidia modules will fail to build, and thus to load. Rebuilding 5.15 against your current packages would fix it.

rado84-github, 2023-08-19:
> unless you meant most changes on newer released wouldn't affect your card.
>
> Shooting in the dark here, but GCC packages were updated recently so if your 5.15 kernel wasn't built against them the [...]

That first line of the quote is what I meant, not that they're dropping support for GTX. My GPU is GTX 1660 Ti.
As for the gcc packages, I've downloaded gcc packages which were released shortly after the 515.57 driver was released. Maybe I should download them released earlier, then?

Tk-Glitch, 2023-08-19:
The Nvidia modules will be rebuilt with dkms (with your currently installed GCC), so what you want is to rebuild the kernel in your current environment for the Nvidia modules and the kernel to match compiler wise.