Frogging-Family/nvidia-all #197: udev rules are broken because missing module is being used for device identification on 545.29.02

valgusk, 2023-11-03:
I think `nvidia-frontend` might be gone from 545.29.02, but it might be only on my computer.
Device files(`/dev/nvidia*`) do not get created as a result. (I noticed it because I have `NVreg_ModifyDeviceFiles=0`)

Files got created successfully after modifying grep commands to use `nvidiactl`, but correct approach would probably be to search for exact match for each specific entry

czapnaper, 2023-11-04:
File was renamed from nvidia-frontend to nvidia, information from issues in [nvidia-kmod-common](https://github.com/negativo17/nvidia-kmod-common/issues/11) and [egl-wayland](https://github.com/NVIDIA/egl-wayland/issues/72#issuecomment-1792908365)

ventureoo, 2024-02-17:
Should be fixed with https://github.com/Frogging-Family/nvidia-all/pull/216ю=

closed 2024-02-18 by Tk-Glitch

valgusk, 2024-02-23:
Did not fit me functionality wise, but I respect the choice for a cleaner approach. 

Just FYI for future reference - it respects `NVreg_ModifyDeviceFiles` and does not create them at all in case of `0`. 

Obnoxiously specific use-case for those interested: I use VFIO forwarding on multi-nvidia-gpu multi-seat environment at random moments and I want device files to be isolated by permissions and static.