Frogging-Family/nvidia-all #210: 535.43.22 is out

hunabku94, 2023-12-21:
Hey, 535.43.22 is out

Vrashabh-Sontakke, 2023-12-22:
Are there any significant performace improvements ?

Tk-Glitch, 2023-12-25:
Added with https://github.com/Frogging-Family/nvidia-all/commit/deb4ac2dc2f019d7d5440304a7d43b6baf14f389

Tk-Glitch, 2023-12-25:
> Are there any significant performance improvements ?

```
December 19th, 2023 - Windows 538.09, Linux 535.43.22

    New:
        [VK_KHR_maintenance6](https://www.khronos.org/registry/vulkan/specs/1.3-extensions/man/html/VK_KHR_maintenance6.html)
        [VK_KHR_video_maintenance1](https://www.khronos.org/registry/vulkan/specs/1.3-extensions/man/html/VK_KHR_video_maintenance1.html)
        Final [Vulkan Video Extensions for Accelerated H.264 and H.265 Encode](https://www.khronos.org/blog/khronos-finalizes-vulkan-video-extensions-for-accelerated-h.264-and-h.265-encode)
            [VK_KHR_video_encode_queue](https://www.khronos.org/registry/vulkan/specs/1.3-extensions/man/html/VK_KHR_video_encode_queue.html)
            [VK_KHR_video_encode_h264](https://www.khronos.org/registry/vulkan/specs/1.3-extensions/man/html/VK_KHR_video_encode_h264.html)
            [VK_KHR_video_encode_h265](https://www.khronos.org/registry/vulkan/specs/1.3-extensions/man/html/VK_KHR_video_encode_h265.html)
    Fixes:
        Improve image creation speed when using VK_NV_extended_sparse_address_space
```

closed 2023-12-25 by Tk-Glitch

ryanmusante, 2023-12-26:
Has anyone tested differences in how vulkan dev handles Wayland vs regular?

Vrashabh-Sontakke, 2023-12-26:
> Has anyone tested differences in how vulkan dev handles Wayland vs regular?

I will test today.

Cardiacman13, 2023-12-26:
The Nvidia driver Vulkan dev version 535.43.22 does not have the flickering/artifacting issues with Wayland that the 545.29.06 may experience. However, it lacks some features introduced with the 545 driver, such as DLSS 3.5 ray reconstruction. Therefore, the 535 Vulkan Dev performs better under Wayland compared to the 545 driver. Personally, I have achieved better results with X11 using the 545 driver. I conducted several tests while [streaming](https://www.twitch.tv/videos/2009126023?filter=archives&sort=time), and X11 with the 545 driver seems to be the optimal compromise at the moment, especially for games with Raytracing and DLSS 3.5.

ryanmusante, 2023-12-26:
I use the /etc/environment variables from below as quoted:
```
SDL_VIDEODRIVER=wayland # Can break some native games
XDG_SESSION_TYPE=wayland
QT_QPA_PLATFORM="wayland;xcb"
MOZ_DBUS_REMOTE=1 # For shared clipboard with Xwayland apps
MOZ_ENABLE_WAYLAND=1
GBM_BACKEND=nvidia-drm
WLR_NO_HARDWARE_CURSORS=1
KITTY_ENABLE_WAYLAND=1
_JAVA_AWT_WM_NONREPARENTING=1
```

https://github.com/ventureoo/nvidia-tweaks