Frogging-Family/nvidia-all #221: Question - Does libxnvctrl have anything to do with nVidia driver ??

Goku-San, 2024-03-04:
Hi, first of all, thanks for the excellent work for easy and complete install of nVidia driver. 
I just installed nVidia driver v.535.161.07-253, because with the latest driver on Arch I had kind of a black screen after suspend.

After installing the driver through this package, I've noticed that 
```zsh
local/libxnvctrl 550.54.14-1
local/nvidia-dkms-tkg 535.161.07-253
```

Should I be worried for ``libxnvctrl`` not being the same version as the driver ??

Thanks..

Mario156090, 2024-08-09:
Hello, why the installer don't compile libxnvctrl and lib32-libxnvctrl too for the version selected?

Mario156090, 2024-09-14:
> Hello, why the installer don't compile libxnvctrl and lib32-libxnvctrl too for the version selected?

@dylanmtaylor

closed 2026-08-19 by damachine