Frogging-Family/nvidia-all #222: Installation is unable to write to boot/efi

sakrayaami, 2024-03-08:
Hello,
I'm experiencing issues on an Arch based distro using the latest kernel and every available version of nvidia-all.
Error I get at the end of the installation is:
`(11/15) Reloading system bus configuration...
(12/15) Update Nvidia modules in initcpio
dracut: Executing: /usr/bin/dracut --kver=6.6.21-1-lts --force
dracut: Can't write to /boot/efi/842c13385be0448d93ab256428336a4b/6.6.21-1-lts: Directory /boot/efi/842c13385be0448d93ab256428336a4b/6.6.21-1-lts does not exist or is not accessible.
dracut: Executing: /usr/bin/dracut --kver=6.7.9-arch1-1 --force
dracut: Can't write to /boot/efi/842c13385be0448d93ab256428336a4b/6.7.9-arch1-1: Directory /boot/efi/842c13385be0448d93ab256428336a4b/6.7.9-arch1-1 does not exist or is not accessible.
dracut: Executing: /usr/bin/dracut --kver=6.7.9-zen1-1-zen --force
dracut: Can't write to /boot/efi/842c13385be0448d93ab256428336a4b/6.7.9-zen1-1-zen: Directory /boot/efi/842c13385be0448d93ab256428336a4b/6.7.9-zen1-1-zen does not exist oris not accessible.
error: command failed to execute correctly
(13/15) Updating the desktop file MIME type cache...
(14/15) Performing snapper post snapshots for the following configurations...
==> root: 906
(15/15) Syncing all file systems...
-> exit cleanup done`

raku-cat, 2024-08-21:
Personally have just been going without the hook and have had no issues, though I have my own hooks for dracut and such.
Here's a patch to quickly drop the lines that I've been using:
```
diff --git a/PKGBUILD b/PKGBUILD
index 7e0f708..350b6ba 100644
--- a/PKGBUILD
+++ b/PKGBUILD
@@ -1891,7 +1891,6 @@ if [ "$_dkms" = "false" ] || [ "$_dkms" = "full" ]; then
                 install -Dm644 /dev/stdin "${pkgdir}/usr/lib/modprobe.d/${pkgname}.conf"
       fi
 
-      install -Dm644 "${srcdir}/nvidia-tkg.hook" "${pkgdir}/usr/share/libalpm/hooks/nvidia-tkg.hook"
   else
       pkgdesc="Full NVIDIA drivers' package for all kernels on the system (drivers and shared utilities and libraries)"
       depends=("nvidia-utils-tkg>=$pkgver" 'libglvnd')
@@ -1922,7 +1921,6 @@ if [ "$_dkms" = "false" ] || [ "$_dkms" = "full" ]; then
                 install -Dm644 /dev/stdin "${pkgdir}/etc/modules-load.d/${pkgname}.conf"
       fi
 
-      install -Dm644 "${srcdir}/nvidia-tkg.hook" "${pkgdir}/usr/share/libalpm/hooks/nvidia-tkg.hook"
   fi
   }
 source /dev/stdin <<EOF
@@ -2062,7 +2060,6 @@ if [ "$_dkms" = "true" ] || [ "$_dkms" = "full" ]; then
       cp -dr --no-preserve='ownership' kernel-dkms "${pkgdir}/usr/src/nvidia-${pkgver}"
 
 
-      install -Dm644 "${srcdir}/nvidia-tkg.hook" "${pkgdir}/usr/share/libalpm/hooks/nvidia-tkg.hook"
 
       install -Dt "${pkgdir}/usr/share/licenses/${pkgname}" -m644 "${srcdir}/${_pkg}/LICENSE"
   fi
```

BernetzederT, 2024-12-19:
I'm facing the same problem with a new Garuda Hyprland installation. Garuda Hyprland is using dracut, but specifically a special version which is 'garuda-dracut-support'. The nvidia-all Script is failing in my case for the same reason as mentioned above: `dracut: Executing: /usr/bin/dracut --kver=6.7.9-zen1-1-zen --force dracut: Can't write to /boot/efi/some_path/6.7.9-zen1-1-zen: Directory /boot/efi/some_path/6.7.9-zen1-1-zen does not exist or is not accessible.` 
/boot/efi/ is not accessible or restricted, it definitely exists.
I've tried to reproduce the workaround by @raku-cat but the code is far too different now in my eyes, as i could say for sure what to change without facing possible serious problems. Is it possible to get a workaround for the actual state the package? or even better, getting a suiting solution for garuda hyprland in general?

raku-cat, 2024-12-19:
I still use that patch to this day so I can assure you it isn't, all it does is remove those 3 instances of installing the pacman hook, you can do it manually if you'd like. 
Like I said in my comment, I have my own hooks that the tkg ones double over and in my case I have my system configured such that not having the hook doesn't break anything, couldn't speak for other systems. I think a configuration option to force dracut/booster/mkinit or to turn off the hooks, or to leave it to the pkgbuilds current auto detections would be ideal if possible.

BernetzederT, 2024-12-21:
> I still use that patch to this day so I can assure you it isn't, all it does is remove those 3 instances of installing the pacman hook, you can do it manually if you'd like. Like I said in my [...]

I've tried it, and it caused no problems, unfortunately it did not solve the issue. It still can't write /boot/efi/ What I also noticed, the script is searching for lib modules for a zen-kernel version which is not installed/outdated.

` makepkg -si
           .---.`               `.---.
        `/syhhhyso-           -osyhhhys/`
       .syNMdhNNhss/``.---.``/sshNNhdMNys.
       +sdMh.`+MNsssssssssssssssNM+`.hMds+
       :syNNdhNNhssssssssssssssshNNhdNNys:
        /ssyhhhysssssssssssssssssyhhhyss/
        .ossssssssssssssssssssssssssssso.
       :sssssssssssssssssssssssssssssssss:
      /sssssssssssssssssssssssssssssssssss/
     :sssssssssssssoosssssssoosssssssssssss:
     osssssssssssssoosssssssoossssssssssssso
     osssssssssssyyyyhhhhhhhyyyyssssssssssso
     /yyyyyyhhdmmmmNNNNNNNNNNNmmmmdhhyyyyyy/
      smmmNNNNNNNNNNNNNNNNNNNNNNNNNNNNNmmms
       /dNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNd/
        `:sdNNNNNNNNNNNNNNNNNNNNNNNNNds:`
           `-+shdNNNNNNNNNNNNNNNdhs+-`
                 `.-:///////:-.`
    
==> WARNUNG: Please make sure you have the corresponding kernel headers package installed for each kernel on your system !

    Which driver version do you want?
    > 1.Vulkan dev: 550.40.82
      2.565 series: 565.77
      3.560 series: 560.35.03
      4.550 series: 550.135
      5.470 series: 470.256.02 (LTS kernel recommended)
      6.Older series
      7.Custom version (396.xx series or higher)
    choice[1-7?]: 2
Build the dkms package or the regular one?
> 1.dkms (recommended)
  2.regular
choice[1-2?]: 1
  -> Checking open driver availability...
  ->  - Open source kernel modules available - 
==> WARNUNG: IT ONLY OFFERS SUPPORT FOR TURING AND NEWER, AND DOESN'T OFFER ALL THE FEATURES OF THE PROPRIETARY ONE.
==> WARNUNG: SLI, G-Sync on notebooks and virtual GPU support are notably missing.
    Do you want to use it instead of the proprietary one?
    > N/y : n
  -> Building driver version 565.77 on branch regular.
  -> Selected driver integrity check behavior (md5sum or SKIP): 48d9a5b5999d0f30ec8326c94b34945e
==> WARNUNG: Die Paket-Gruppe wurde bereits gebaut, installiere existierende Pakete...
==> Installiere Paketgruppe nvidia-dkms-tkg mit pacman -U...
[sudo] Passwort für germanikus:              
Pakete werden geladen …
Warnung: nvidia-dkms-tkg-565.77-263 ist aktuell -- Reinstalliere
Warnung: nvidia-utils-tkg-565.77-263 ist aktuell -- Reinstalliere
Warnung: lib32-nvidia-utils-tkg-565.77-263 ist aktuell -- Reinstalliere
Warnung: opencl-nvidia-tkg-565.77-263 ist aktuell -- Reinstalliere
Warnung: lib32-opencl-nvidia-tkg-565.77-263 ist aktuell -- Reinstalliere
Warnung: nvidia-settings-tkg-565.77-263 ist aktuell -- Reinstalliere
Warnung: nvidia-egl-wayland-tkg-565.77-263 ist aktuell -- Reinstalliere
Abhängigkeiten werden aufgelöst …
Nach in Konflikt stehenden Paketen wird gesucht …

Paket (7)                Alte Version  Neue Version  Netto-Veränderung

lib32-nvidia-utils-tkg   565.77-263    565.77-263             0,00 MiB
lib32-opencl-nvidia-tkg  565.77-263    565.77-263             0,00 MiB
nvidia-dkms-tkg          565.77-263    565.77-263             0,00 MiB
nvidia-egl-wayland-tkg   565.77-263    565.77-263             0,00 MiB
nvidia-settings-tkg      565.77-263    565.77-263             0,00 MiB
nvidia-utils-tkg         565.77-263    565.77-263             0,00 MiB
opencl-nvidia-tkg        565.77-263    565.77-263             0,00 MiB
[... 113 lines cut ...]
/etc/issue.pacnew
/etc/locale.gen.pacnew
/etc/pacman.conf.pacnew
/etc/makepkg.conf.d/rust.conf.pacnew
/etc/pacman.d/mirrorlist.pacnew
/etc/updatedb.conf.pacsave
Please check and merge
(12/16) Reloading system bus configuration...
(13/16) Update Nvidia modules in initcpio
dracut[I]: Executing: /usr/bin/dracut --kver=6.12.4-arch1-1 --force
dracut[F]: Can't write to /boot/efi/992238dd28d5465e9a85fc47122d3a60/6.12.4-arch1-1: Directory /boot/efi/992238dd28d5465e9a85fc47122d3a60/6.12.4-arch1-1 does not exist or is not accessible.
dracut[I]: Executing: /usr/bin/dracut --kver=6.12.4-zen1-1-zen --force
dracut[F]: Can't write to /boot/efi/992238dd28d5465e9a85fc47122d3a60/6.12.4-zen1-1-zen: Directory /boot/efi/992238dd28d5465e9a85fc47122d3a60/6.12.4-zen1-1-zen does not exist or is not accessible.
dracut[I]: Executing: /usr/bin/dracut --kver=6.12.5-273-tkg-eevdf --force
dracut[F]: Can't write to /boot/efi/992238dd28d5465e9a85fc47122d3a60/6.12.5-273-tkg-eevdf: Directory /boot/efi/992238dd28d5465e9a85fc47122d3a60/6.12.5-273-tkg-eevdf does not exist or is not accessible.
dracut[I]: Executing: /usr/bin/dracut --kver=6.12.6-lqx1-1-lqx --force
dracut[F]: Can't write to /boot/efi/992238dd28d5465e9a85fc47122d3a60/6.12.6-lqx1-1-lqx: Directory /boot/efi/992238dd28d5465e9a85fc47122d3a60/6.12.6-lqx1-1-lqx does not exist or is not accessible.
Fehler: Befehl konnte nicht korrekt ausgeführt werden
(14/16) Updating the desktop file MIME type cache...
(15/16) Performing snapper post snapshots for the following configurations...
==> root: 443
(16/16) Syncing all file systems...
  -> exit cleanup done`

the drivers seem to be installed so far, but I'm not sure, if there are any negative sideffects. So far I can only see, that the performance is not, what i should be, but it's far better than before.

raku-cat, 2024-12-21:
The errors about old zen kernel versions are from dkms and are unrelated to this package entirely.
I suggest checking out https://wiki.archlinux.org/title/Dynamic_Kernel_Module_Support though they are harmless errors so long as your system is still booting.
The dracut ones are because the pacman hooks are still around, you would've had to purge the nvidia-all package and then rebuild it with my patch for the desired effect. 
To fix manually now, just remove `/usr/share/libalpm/hooks/nvidia-tkg.hook` and either rebuild or  `pacman -U` your existing latest packages your built.
None of this would make any difference to performance really, the only "adverse affect" would be your system booting or not.

BernetzederT, 2024-12-29:
Indeed it works perfectly, thanks a lot for the help :)

raku-cat, 2025-08-13:
With the merging of https://github.com/Frogging-Family/nvidia-all/pull/302 you can now workaround this with `_disable_libalpm_hook="true"` in your `customization.cfg`, no patch needed! 
I have to assume this error is simply from a conflict caused by running two hooks however OP would have to confirm if they had an existing hook, certainly for me and the other commenter had one though, else removing the hook would've caused breakage.