Frogging-Family/nvidia-all #236: Can't install 550.40.07 on Kernel 6.8.9

Rune580, 2024-05-09:
I know it's a weird combo, but that's the last version of 550 that still worked with Unity Editor 2022+.

[make.log](https://github.com/Frogging-Family/nvidia-all/files/15264401/make.log)

Related:
- https://issuetracker-mig.prd.it.unity3d.com/issues/linux-nvidia-vulkan-silent-crash-when-opening-windows-on-a-vulkan-project-with-nvidia-550-drivers
- https://forums.developer.nvidia.com/t/550-47-unity-editor-on-linux-crashes-when-using-vulkan-graphics-backend-with-nvidia-550-drivers/288990

ProgAchr, 2024-05-11:
I updated my archlinux install, my system boot but KDE run in low resolution, and nvidia-settings is empty (driver not recognized)
I tried to downgrade from kernel 6.8.9 with no luck. I had to wipe my system and re-install. 
I found others having the same problem with the latest kernel in the archlinux forums:
System stopped booting with non-latest NVIDIA drivers https://bbs.archlinux.org/viewtopic.php?pid=2170207#p2170207
nvidia-390xx-dkms build failure https://bbs.archlinux.org/viewtopic.php?id=295600

mkfyi, 2024-05-21:
Same with 525xx and  kernel `6.9.1` and `6.8.9`, my `make.log` throws the same warnings as for @Rune580. Had no issues before upgrading my system from `6.8.2`.

Tk-Glitch, 2024-05-21:
Please give the new option a try by setting `_gcc14_fix="false"` to `_gcc14_fix="true"`. Considering the logs it should work.

ProgAchr, 2024-05-24:
i set _gcc14_fix="true" in customization.cfg, driver 525.147.05 work correctly with archlinux kernel 6.8.9 now, issu solved.

Tk-Glitch, 2024-05-24:
Great news! Closing.